ASCOMInitiative/ASCOMPlatform #1: ASCOM header files

ThomasRuppel, 2018-06-07:
Hello,

I would like to integrate the ASCOM dlls into a Matlab script (by using "loadlibrary(dll-file,header-file)" and I look for the necessary headers files. Unfortunately, in the ASCOM developer Tools only XML-files are included. Are C Header files available for the ASCOM dlls?

Best regards,
Thomas

NameOfTheDragon, 2018-07-03:
Most of the platform is written using .NET so the dlls are .NET assemblies and there are no header files (.NET doesn't use them). However, many of the ASCOM components are also usable as language independent COM objects. Loading as COM components might be a better approach for you.

Peter-Simpson, 2018-07-05:
Hi Thomas,

Matlab can access .NET assemblies directly and the help for this is here: [Use .NET Assemblies in Matlab](https://uk.mathworks.com/help/matlab/getting-started.html).

Regards, Peter

Peter-Simpson, 2018-08-03:
Closed because question is answered and no further response from Thomas.

closed 2018-08-03 by Peter-Simpson