ASCOMInitiative/ASCOMPlatform #2: Git users not showing up correctly in the commit history

NameOfTheDragon, 2018-07-03:
All of my commits are showing up as anonymous in the commit history. My Git user ID is "NameOfTheDragon" and I would like my historical commits to be associated with that user ID so that I'm correctly identified as the author of my work.

Peter-Simpson, 2018-07-05:
Hi Tim,

I checked ASCOM.Powershell.Wix as an example and you are attributed as Tim Long, see below:

[image]

Can you give me an example of an "anonymous" commit that I can follow up?

Thanks, Peter

Peter-Simpson, 2018-07-05:
Hi Tim,

Grasping at straws here, were you using a different email address in the past to the one you currently use?

Regards, Peter

NameOfTheDragon, 2018-07-05:
"Anonymous" was probably a bad choice of word. What I mean is that it has not associated my commits with my GitHub user ID (NameOfTheDragon). According to GitHub, this is supposed to work by matching email address but it obviously hasn't matched any of the 3 addresses I have registered with GitHub. The commits therefore do not show up in my GitHub statistics, which are important to me for a variety of reasons.

I reached out to GitHub support to see if this could be fixed retrospectively, and they sent me this:

> That repository administrator would need to change the name and/or email address recorded in
> existing commits by rewriting the entire history of the repository. 
> Our team wrote a guide for doing this here:
> https://help.github.com/articles/changing-author-info/

Peter-Simpson, 2018-07-06:
Hi Tim,

I have these emails associated with my GitHub account:
[image]
and do get attribution. Your equivalent address is shown here:

[image]

Is TimLong@60bb46e1-9293-8f4a-b0db-2b7d974c5fa5 associated with your GitHub ID?

I also found this set of GitHub requirements in order for commits to be included in attribution:

[image]

It looks to me like you meet the first group of requirements, but may be missing out on one from the second group? Perhaps its worth "Starring" the Platform to see if this makes a difference?

Regards, Peter

NameOfTheDragon, 2018-07-06:
Those strange looking "addresses" are clearly something made up by the importer and are not actual email addresses, although GitHub did allow me to add it to my account. That hasn't made any difference though. If you look at the project summary, it says there are only 2 contributors, which is an unfortunate rewriting of history.

If I fork the repository, then that changes to 3 contributors although that's just an ownership thing and it still doesn't associate me with my old commits, of which there are many hundreds.

Although I've never done it on GitHub, in my experience it is necessary to create a user mapping file that maps SVN users to Git style users prior to importing the repository. At least that's how I've done it in the past on BitBucket. It may be too late to go back and do that now, which is unfortunate for people like me who will be whitewashed out of the history. I am kicking myself because I had a fully mapped Git mirror of the SourceForge repository for many months on BitBucket, with all the contributors intact. When I saw you had started using Git, I deleted it! 

I believe the only two options at this point are to (1) re-do the import from SVN with a user mapping file, or (2) to re-write the history as detailed in the GitHub howto document that I previously linked to. The  longer you leave it, the less feasible these options become.

Peter-Simpson, 2018-08-03:
Hi Tim,

Your attribution is now showing up as requested, is it OK to close this issue?

Regards, Peter

[image]

NameOfTheDragon, 2019-01-03:
Apologies for delay in responding. Thank you for attending to this, it is much appreciated. Marking as closed.

closed 2019-01-03 by NameOfTheDragon