ASCOMInitiative/ASCOMPlatform #3: NOVAS31.deltaT throwing "Not a legal OleAut date"

isbeorn, 2018-07-16:
Hi,

in ASCOM version 6.4. this method is not working anymore. This was working fine in version 6.3 and below.
When calling it with a julian date retrived via NOVAS31.JulianDate() this method is throwing "Not a legal OleAut date"

at System.DateTime.DoubleDateToTicks(Double value)
   at ASCOM.Astrometry.EarthRotationParameters.DeltaT(Double RequiredDeltaTJulianDateUTC) in C:\ASCOM Build\Export\ASCOM.Astrometry\ASCOM.Astrometry\EarthRotationParameters.vb:line 493

Did the input for this method change, or what's wrong here?

Regards,
Stefan

isbeorn, 2018-07-16:
It looks like the problem occurs only for the default datetime (which is 01/01/0001). When i set it to a current date it works again.

Peter-Simpson, 2018-08-03:
Hi Stefan,

Yes, the implementation of DeltaT has changed in 6.4 to enable the Platform to provide the latest delta T and leap second information through Internet download. The revised implementation uses the DateTime.FromOADate() function as an efficient way to convert a Julian date to a .NET DateTime date. 

The exception you are seeing is triggered by the Julian date that you are supplying, 1 January 0001, which is earlier than the earliest valid OLEAutomation date - midnight, 1 January 0100.

Given the lack of precise earth rotation data for this epoch, any calculated value of DeltaT will be meaningless and consequently I don't think that there is anything I can do to make this function useful on 1 January 0001.

I appreciate that the error message isn't helpful and will revise the Platform code to throw a more helpfully worded ASCOM exception.

I am assuming here that you don't have a real need for the DeltaT value on 1 January 0001 and that the call using this value was unintentional. If this is the case, the best approach would be to revise your code so as not to call Novas31.DeltaT with this value.

Regards, Peter

JPhilC, 2019-01-06:
Peter,

I am also hitting this problem re-running some unit tests that did work prior to 6.4. I am using a valid date. To help you track the problem down here is a little unit test to replicate the problem.

```c-sharp
using System;
using ASCOM.Astrometry.Transform;
using Microsoft.VisualStudio.TestTools.UnitTesting;

namespace LunaticAstroEQ.Tests
{
   [TestClass]
   public class TransformTest
   {
      [TestMethod]
      public void TestMethod1()
      {
         using (Transform transform = new Transform())
         {
            double longitude = 1.3223722222222223;
            transform.SiteLatitude = 52.65177222222222;
            transform.SiteLongitude = longitude;
            transform.SiteElevation = 175.5;
            transform.JulianDateUTC = LocalApparentSiderealTime(longitude, new DateTime(2017, 03, 28, 21, 0, 0));
            transform.SetTopocentric(0.13166666666666665, 29.0);
            double alt = transform.ElevationTopocentric;
            double az = transform.AzimuthTopocentric;
            Assert.AreEqual(alt, 10.0); 
         }
      }

      private double LocalApparentSiderealTime(double longitude, DateTime localTime)
      {
         // get greenwich sidereal time: https://en.wikipedia.org/wiki/Sidereal_time
         //double siderealTime = (18.697374558 + 24.065709824419081 * (utilities.DateUTCToJulian(DateTime.UtcNow) - 2451545.0));

         // alternative using NOVAS 3.1
         double siderealTime = 0.0;
         using (var novas = new ASCOM.Astrometry.NOVAS.NOVAS31())
         {
            var jd = localTime.ToUniversalTime().ToOADate() + 2415018.5;      // Taken from ASCOM.Util.DateUTCToJulian
            novas.SiderealTime(jd, 0, novas.DeltaT(jd),
                ASCOM.Astrometry.GstType.GreenwichApparentSiderealTime,
                ASCOM.Astrometry.Method.EquinoxBased,
                ASCOM.Astrometry.Accuracy.Reduced, ref siderealTime);
         }
         // allow for the longitude
         siderealTime += longitude / 360.0 * 24.0;
         // reduce to the range 0 to 24 hours
         siderealTime = siderealTime % 24.0;
         return siderealTime;
      }
   }
}

```
The exeption will be thrown on the line `double alt = transform.ElevationTopocentric;`

Kind regards,

Phil

JPhilC, 2019-01-06:
Just looking at the exception call stack it is the following line that is causing the problem.
`                LogDebugMessage("DeltaUT1(JD)", String.Format("Automatic DeltaUT1 is required for Julian date: {0} ({1})", RequiredDeltaUT1JulianDateUTC,
                                                       DateTime.FromOADate(RequiredDeltaUT1JulianDateUTC - OLE_AUTOMATION_JULIAN_DATE_OFFSET).ToString(DOWNLOAD_TASK_TIME_FORMAT)))
`
Line 649 in EarthRotationParameters.vb. I think you have an error in your debug message logging.

The call stack is:
```
   at System.DateTime.DoubleDateToTicks(Double value)
   at System.DateTime.FromOADate(Double d)
   at ASCOM.Astrometry.EarthRotationParameters.DeltaUT1(Double RequiredDeltaUT1JulianDateUTC) in C:\ASCOM Build\Export\ASCOM.Astrometry\ASCOM.Astrometry\EarthRotationParameters.vb:line 649
   at ASCOM.Astrometry.AstroUtils.AstroUtils.DeltaUT(Double JulianDate) in C:\ASCOM Build\Export\ASCOM.Astrometry\ASCOM.Astrometry\AstroUtils.vb:line 400
   at ASCOM.Astrometry.Transform.Transform.TopoToJ2000() in C:\ASCOM Build\Export\ASCOM.Astrometry\ASCOM.Astrometry\Transform.vb:line 582
   at ASCOM.Astrometry.Transform.Transform.Recalculate() in C:\ASCOM Build\Export\ASCOM.Astrometry\ASCOM.Astrometry\Transform.vb:line 653
   at ASCOM.Astrometry.Transform.Transform.get_ElevationTopocentric() in C:\ASCOM Build\Export\ASCOM.Astrometry\ASCOM.Astrometry\Transform.vb:line 438
   at LunaticAstroEQ.Tests.TransformTest.TestMethod1() in C:\Users\phil\source\repos\LunaticAstroEQ\LunaticAstroEQ.Tests\UnitTest1.cs:line 21
```

Hope this helps.

Peter-Simpson, 2019-04-04:
Closed due to lack of further activity

closed 2019-04-04 by Peter-Simpson