ASCOMInitiative/ASCOMPlatform #4: Windows (10) Defender reported some files are infected by Trojan:Win32/Emali.A!cl

jin-young, 2018-10-01:
Affected items:
containerfile: D:\download\ASCOMPlatform64.exe

file: D:\download\ASCOMPlatform64.exe->(7zSfx)->ASCOMPlatform64.res->RemoveASCOM.exe

Affected items:
containerfile: C:\ProgramData\miaC621.tmp\ASCOMPlatform64.res

file: C:\ProgramData\miaC621.tmp\ASCOMPlatform64.res->RemoveASCOM.exe
file: C:\Users\mayo\AppData\Local\Temp\mia1\RemoveASCOM.exe

Above is what I got from Windows Defender during installing 6.4 SP1. Is it false alarm?

Peter-Simpson, 2018-10-01:
Hi,

Yes, this is a false alarm. I've run all three files through VirusTotal and only the Microsoft scanner has an issue with RemoveASCOM, it is passed by 59 other anti-virus programs including McAfee and Symantec, Zone Alarm and Trend.

Unfortunately we have found that it is quite usual to receive a few reports like this just after we make a new release.

Here are the links to VirusTotal reports for the three files, I believe that the small number of issues flagged are all false positives.

[ASCOM Platform](https://www.virustotal.com/#/file/af579031f8fd891ee3026579d2a2249d2aa63467f07e19f48f12049e9abc5a66/detection)
[ASCOM Developer Components](https://www.virustotal.com/#/file/302f82c70379479e9f582b55d82f70fd122b1664da8f49cccc8c2eb07f588d59/detection)
[Remove ASCOM Zip File](https://www.virustotal.com/#/file/80e929972e0073ad2d03f1b18a2586dc032aa52af401929685c8894358d345c0/detection)

Regards, Peter

Peter-Simpson, 2019-04-04:
Closing because no further response.

closed 2019-04-04 by Peter-Simpson