ASCOMInitiative/ASCOMPlatform #7: Skeleton driver template issue

astroman133, 2019-05-02:
In a C# driver that is generated from the ASCOM device driver template a variable is created to hold a static instance of the TraceLogger object that is used throughout the driver for logging driver activity as specified by the author. This is the C# code:

        /// <summary>
        /// Variable to hold the trace logger object (creates a diagnostic log file with information that you specify)
        /// </summary>
        internal static TraceLogger tl;

Also generated is a Dispose method as follows:

        public void Dispose()
        {
            // Clean up the tracelogger and util objects
            tl.Enabled = false;
            tl.Dispose();
            tl = null;
            utilities.Dispose();
            utilities = null;
            astroUtilities.Dispose();
            astroUtilities = null;
        }

Notice that Dispose cleans up and disposes the TraceLogger object. The issue is that 'tl' is static so if there are two instances of a driver in a local server they both share the same trace logger. And of one of those instances is disposed by its client, the static trace logger is also disposed. Then when the remaining instance of the driver attempts to write to the trace logger, a null reference exception is generated.

The fix may be as simple as removing the static keyword from the definition of the TraceLogger instance variable.

Peter-Simpson, 2019-05-29:
Thanks, I've done as suggested. Each driver instance will now have its own trace logger that it can dispose of when appropriate. This follows the model used in the VB template.

The change will appear in the next Platform release so I'm closing this issue.

closed 2019-05-29 by Peter-Simpson

astroman133, 2020-02-03:
Based on the above changes the 6.5 pre-release templates now generate compile errors because the SetupDialogForm still references the trace log field as a static field. If the only way that the form object can be instantiated is from within the driver then the easiest fix may be to pass the driver instance into the SetupDialogForm constructor.

Peter-Simpson, 2020-02-03:
Oops... good catch, I'll send you a revised version shortly.

commit 583ad62f (2020-02-03): Fix issue arising from changes to fix #7.