ASCOMInitiative/ASCOMPlatform #8: Conform suggested enhancements

astroman133, 2019-05-07:
Was running Conform and saw some minor enhancements that might be of value for a testing telescope driver:

Attempt to write an unsupported TrackingRate value should raise an error (not currently tested)
Reading TargetRightAscension and TargetDeclination after valid values have been written (not currently tested).

Peter-Simpson, 2019-05-29:
Thanks, these are good ideas. I've added TrackingRate tests to Conform and also fixed the Telescope Simulator so that it will reject invalid TrakingRate values. Up until now it accepted invalid values without complaint.

The simulator change will be in the next Platform release and the Conform change will be in it's next release.

Conform already checks that values written to TargetRightAscension and TargetDeclination read back correctly, I think this suggestion is already covered.

Is this now OK to close?

Thanks, Peter

Peter-Simpson, 2019-06-10:
The revised Conform version has been released so I'm closing this suggested as it's been delivered.

Regards, Peter

closed 2019-06-10 by Peter-Simpson