ASCOMInitiative/ASCOMPlatform #11: Incorrect Conform error reported

astroman133, 2019-05-27:
Testing the ASCOM.Simulator.Focuser with Conform version 6.4.1.2694 reports the following error:

06:34:27.948 Move - TempComp True              ERROR    TempComp is True but no exception is thrown by the Move Method - See Focuser.TempComp entry in Platform help file

The simulator reportedly supports IFocuserV3 (InterfaceVersion = 3). This should mean that focuser movement when TempComp is true is allowed and no exception would be expected.

This seems like a bug in Conform.

Thanks!

Peter-Simpson, 2019-05-28:
Hi Rick,

 

I’m not seeing that here, I’ve included the Conform output below. 6.4.1.2748 is the Platform version, the Conform version is 6.4.63.4 in the example below. Can you check which Conform version you are running?

 

Regards, Peter

 

Start-up ASCOM Device Conformance Checker - 64bit mode

Start-up ASCOM Platform 6.4 SP1 6.4.1.2748

 

ConformanceCheck ASCOM Device Conformance Checker Version 6.4.63.4, Build time: 28/05/2019 15:02:06

ConformanceCheck Running on: ASCOM Platform 6.4 SP1 6.4.1.2748

ConformanceCheck Driver ProgID: ASCOM.Simulator.Focuser

Error handling 

Error number for "Not Implemented" is: 80040400

Error number for "Invalid Value 1" is: 80040404

Error number for "Value Not Set 1" is: 80040402

Error number for "Value Not Set 2" is: 80040403

Error messages will not be interpreted to infer state.

15:02:42.854 Driver Access Checks              OK       

15:02:43.480 AccessChecks                      OK       Successfully created driver using late binding

15:02:43.623 AccessChecks                      OK       Successfully connected using late binding

15:02:43.626 AccessChecks                      INFO     The driver is a .NET object

15:02:43.629 AccessChecks                      INFO     The AssemblyQualifiedName is: ASCOM.Simulator.Focuser, ASCOM.Simulator.Focuser, Version=6.4.0.0, Culture=n

15:02:43.632 AccessChecks                      INFO     The driver implements interface: ASCOM.DeviceInterface.IFocuserV3

15:02:43.634 AccessChecks                      INFO     The driver implements interface: System.IDisposable

15:02:44.353 AccessChecks                      INFO     Device does not expose IFocuser interface

15:02:44.511 AccessChecks                      INFO     Device does not expose IFocuserV2 interface

15:02:44.648 AccessChecks                      INFO     Device exposes IFocuserV3 interface

15:02:44.936 AccessChecks                      OK       Successfully created driver using driver access toolkit

15:02:44.951 AccessChecks                      OK       Successfully connected using driver access toolkit

15:02:45.086 AccessChecks                      OK       Successfully disconnected using driver access toolkit

Conform is using CreateObject to get a Focuser object

15:02:45.207 ConformanceCheck                  OK       Driver instance created successfully

15:02:45.277 ConformanceCheck                  OK       Connected OK

Common Driver Methods 

15:02:45.335 InterfaceVersion                  OK       3

15:02:45.365 Connected                         OK       True

15:02:45.393 Description                       OK       ASCOM Focuser Simulator Driver

15:02:45.419 DriverInfo                        OK       Focuser Simulator Driver

15:02:45.446 DriverVersion                     OK       6.4

15:02:45.472 Name                              OK       ASCOM.Simulator.Focuser

15:02:45.498 CommandString                     INFO     Conform cannot test the CommandString method

15:02:45.502 CommandBlind                      INFO     Conform cannot test the CommandBlind method

15:02:45.505 CommandBool                       INFO     Conform cannot test the CommandBool method

15:02:45.509 Action                            INFO     Conform cannot test the Action method

15:02:45.513 SupportedActions                  OK       Driver returned an empty action list

Properties 

15:02:45.615 Absolute                          OK       True

15:02:45.620 IsMoving                          OK       False

15:02:45.624 MaxStep                           OK       5000

15:02:45.628 MaxIncrement                      OK       5000

15:02:45.631 Position                          OK       5000

15:02:45.635 StepSize                          OK       20

15:02:45.639 TempCompAvailable                 OK       True

15:02:45.643 TempComp Read                     OK       False

15:02:45.647 TempComp Write                    OK       Successfully turned temperature compensation on

15:02:45.650 TempComp Write                    OK       Successfully turned temperature compensation off

15:02:45.654 Temperature                       OK       3.78

Methods 

15:02:45.693 Halt                              OK       Focuser halted OK

15:02:45.701 Move - TempComp False                      Moving to position: 4500

15:02:47.308 Move - TempComp False                      Asynchronous move found

15:02:47.320 Move - TempComp False             OK       Absolute move OK

15:02:47.327 Move - TempComp False             INFO     Returning to original position: 5000

15:02:48.936 Move - TempComp True V3                    Moving to position: 4500

15:02:50.554 Move - TempComp True V3                    Asynchronous move found

15:02:50.558 Move - TempComp True V3           OK       Absolute move OK

15:02:50.566 Move - TempComp True V3           INFO     Returning to original position: 5000

15:02:52.172 Move - To 0                                Moving to position: 0

15:02:59.647 Move - To 0                                Asynchronous move found

15:02:59.655 Move - To 0                       OK       Moved to 0

15:02:59.663 Move - Below 0                             Moving to position: -10

15:03:00.736 Move - Below 0                             Asynchronous move found

15:03:00.740 Move - Below 0                    OK       Moved to 0

15:03:00.745 Move - To MaxStep                          Moving to position: 5000

15:03:08.220 Move - To MaxStep                          Asynchronous move found

15:03:08.225 Move - To MaxStep                 OK       Moved to 5000

15:03:08.230 Move - Above Maxstep                       Moving to position: 5010

15:03:09.301 Move - Above Maxstep                       Asynchronous move found

15:03:09.306 Move - Above Maxstep              OK       Moved to 5000

Conformance test complete 

 

No errors, warnings or issues found: your driver passes ASCOM validation!!

Peter-Simpson, 2019-05-28:
Thanks Rick, I'll close the issue.

Regards, Peter

closed 2019-05-28 by Peter-Simpson