ASCOMInitiative/ASCOMPlatform #12: comport

esittler, 2019-06-01:
ASCOM doesn't recognize the usb port to my telescope is connected to. My Dell computer does recognize it. I have new SynScam V5 Controller which has USB port so connection cable issues should not apply unless ASCOM and SynScan V5 controller are not compatible. Could it be a driver problem? If so, where do I get the driver? Ed

Peter-Simpson, 2019-06-10:
Hi Ed,

Have you managed to fix this or are you still stuck?

I presume that the SynScan 5 is creating a virtual COM port and we have previously come across devices that are not identified by the .NET "Installed COM Ports" routine on which the Platform relies.

In the first instance I suggest connecting the SynScan, running an ASCOM Diagnostic and posting the resulting log file, which will contain information on the COM ports that can be seen.

The Diagnostic utility is in your  "Start Menu \ Programs \ ASCOM Platform 6" folder.

Regards, Peter

esittler, 2019-06-15:
Peter, Working with Orion Tech support I was able to get their Synscan
Update to work. The manual doesn't say it, but first the Synscan V Hand
controller must be disconnected from everything which includes the
telescope motor controller. You then hold the 0-8 buttons down together and
then insert the usb connector and then the Synscan Update will appear on
the screen. Then using their Synscan Firmware Loader Ver. 3.3, have Enforce
database update unclicked and then click on HC Version it came back with
software version on the controller which was their latest version but most
importantly showed my Dell PC laptop was communicating with the controller.
So cables etc are not the problem. Then I got in the controller's "UTILITY
FUNCTION\ PC Direct Mode". Manual says under this mode the controller
becomes a repeater between the PC and the motor controller inside the
telescope mount. I then use the Astrophotography Program, which uses ASCOM,
to both control my camera and the telescope motors. Camera works ok. But,
when I get in the Gear Tool, I select Telescope or something like that,
ASCOM panel comes, I select Sky Telescope etc but doesn't recognize in this
case com9. I will try your ASCOM diagnostics before but will try again
since with new controller V I have usb connection and we know it works. Ed

brujah1980, 2020-01-23:
Any news on this?  I am having the same issue.  I was able to update my HC to the newest firmware and still no connection from ASCOM platform 6.  I have installed the ASCOM synscan drivers but it keeps installing the standard Prolific USB drivers, which I feel may be interfering with the ASCOM ones.  Any help would be greatly appreciated.

Peter-Simpson, 2020-01-25:
Hi brujah1980,

This Platform issues log is to deal with faults in the ASCOM Platform itself and so far I have no evidence that the Platform itself is mis-behaving because Ed never posted the Diagnostics log that I requested.

I've re-read the posts here but it will help me if we can confirm your issue. Is it that the COM port exposed by the Synscan hand controller is not visible in the driver configuration dialogue, or is it something else?

If the former, please can you:
- Connect the Synscan to your PC
- Power up scope and Synscan
- Start the ASCOM Diagnostics tool (needs admin permission), which is in the Start/ASCOM Platform 6 folder
- In the Diagnostics Tools menu there is a "List Available COM Ports" option, does the Synscan COM port show up?
- Click the Run Diagnostics button. This will take several minutes to run.
- Attach the Diagnostics log to a new post. The log location is shown on the Diagnostic tool main window.

Regards, Peter

esittler, 2020-01-25:
Peter, I’m not in a position to respond at this time but I ran out of time
by end of June for trip to Chile for Solar Eclipse so had to manually
control the motors with the SynScan controller and worked okay and got
great pictures but will want to control the motors in the future via APT.
Ed

Peter-Simpson, 2020-01-28:
Thanks for the update Ed. 
There is a Platform configuration mechanic to force COM ports to appear even when they can not be auto-discovered. I've created an FAQ covering this here: [How to force a COM port to appear](https://ascomtalk.groups.io/g/Help/wiki/How-to-force-a-COM-port-to-appear-in-the-list-of-available-COM-ports). 
I'm going to close this issue because I don't think it is one that can be addressed through a Platform update without further information.

closed 2020-01-28 by Peter-Simpson

brujah1980, 2020-02-07:
Sorry for the delayed response, been a bit busy lately.  I found my issue, hope it helps OP as well.  The synscan v5 handset has built n USB, but the instructions are all for the v4 HC.  With the v5, when you put the HC in PC Direct mode, the ASCOM compatible port is not recognized.  What I recently found is that when you plug it in without PC Direct Mode, ASCOM Chooser will recognize the port as one with a compatible product.  I am happily connected to my telescope mount and working on the little orientation settings in Stellarium.  Thank you.

Peter-Simpson, 2020-02-08:
Great, I'm glad that you have found a solution!

Best wishes, Peter

esittler, 2020-02-09:
This is great news I’ll have to try it out. Ed