ASCOMInitiative/ASCOMPlatform #13: ASCOM Dome Control Panel Run-time error

jp-technology, 2019-06-19:
Hi, 

When using Sequence Generator PRO, the Observatory will occasionally disconnect and a message box titled "ASCOM Dome Control Panel" will appear containing the following lines of text:
_____________________________________________

Run-time error '-2147352567 (80020009)':

Unable to complete command
_____________________________________________

Clicking OK causes the ASCOM Dome Control Panel to disappear from the screen. I can then click reconnect in SGP and the control panel will restart, but often the dome will likely have closed before I've realized the ASCOM Dome CP crashed and any exposures in progress need to be restarted.

The dome being controlled is a MaxDome II

If there are any system logs that you can direct me to I would be happy to provide them.

Thanks,

John

astroman133, 2019-07-10:
You should probably post this issue on the ASCOM-Talk forum. Then others can provide troubleshooting advice. If your issue turns out to be something that is addressable within the Platform or the Dome Control Panel it would be appropriate to post the details of the issue here.

NameOfTheDragon, 2019-12-28:
My view on this is that there's not really any point in having 3 different scope-dome hubs in the platform. Given that both the Dome Control Panel and POTH predate the changes to `SideOfPier`, neither of them really work properly any more and they are effectively not maintainable, being written in VB6. In due course, it might be an idea to retire POTH and Dome Control Panel and replace both with the newer Device Hub, which does work correctly with the new definition of `SideOfPier`

Peter-Simpson, 2020-01-27:
I'm closing this issue because the new Device Hub will be added to the Platform 6.5 release that will be delivered in the next one to two months.

closed 2020-01-27 by Peter-Simpson