ASCOMInitiative/ASCOMPlatform #14: Annunciator control doesn't always display correctly

NameOfTheDragon, 2019-07-24:
The [Annunciator control](https://github.com/ASCOMInitiative/ASCOMPlatform/blob/master/ASCOM.Controls/ASCOM.Controls/Annunciator.cs) doesn't always display correctly. There are some corner cases when the control's on/off state or cadence are changed that result in the control being invisible or not displaying in the correct cadence. Improvement is desirable.

The LED control appears to be affected in a similar way.

This would be a great first issue for someone new to the project to take a crack at.

NameOfTheDragon, 2019-12-28:
This is now obsolete because the controls have been spun out into a separate NuGet package where I can maintain it at my own speed. They can therefore be removed from the platform distribution in due course. It might be useful to signpost developers to the NuGet package because the controls are styled specifically for use in ASCOM drivers. I have added the "won't fix" label to indicate that the code in the platform will not be maintained. For reference, the NuGet package is at https://www.nuget.org/packages/TA.WinForms.Controls/

Should we close this issue or leave it around as a reminder?

Peter-Simpson, 2020-01-27:
Hi Tim,

I've just tried to use your new controls package in the Telescope Simulator Server, which is targeted at the Platform default of .NET 4.0 Client, but it won't install because the package is only targeted at a minimum of 4.7.2.

Can you create a version targeted at 4.0 Client?

Regards, Peter

NameOfTheDragon, 2020-01-28:
That should be possible, so yes. I always target the current release when creating a new project so I would have targeted 4.7.2 by default. I'll look at multi-targeting the project.

NameOfTheDragon, 2020-01-28:
@Peter-Simpson Unfortunatelt after looking into it, it turns out that this is not going to be possible. The earliest framework version I could possibly go to is 4.5. The reason is that I'm relying on some features of the Task Parallel Library (`Task<T>`) that were not introduced until .NET 4.5. Specifically, Task.ConfigureAwait().

This has implications for how task continuations are queued and is required for ensuring correct thread management in a WinForms environment. Without this, it is possible for clients of the library to get into a deadlock situation. Some background information on this from Microsoft's guidelines for `async`/`await`: [Configure Context](https://medium.com/bynder-tech/c-why-you-should-use-configureawait-false-in-your-library-code-d7837dce3d7f)

I could ignore the deadlock issue and not use `ConfigureAwait()` but I'd rather not put out a version of my library with that vulnerability in it.

The problem with not keeping up with framework versions is that you inevitably encounter an increasing number of libraries that can no longer support you. My suggestion would be to retarget the simulator projects to .NET 4.7.2. This should not cause any issues as it is a superset of .NET 4.0 and uses the same CLR version. I regularly develop drivers that target .NET 4.7+ and there are no issues with doing this that I am aware of.

Peter-Simpson, 2020-01-28:
> Reply to: ASCOMInitiative/ASCOMPlatform <reply@reply.github.com>

Hi Tim,

 

I agree it’s better to up the Platform minimum level to 4.7.2 as you suggest. Could you respond to this effect in my minimum Framework version topic on the Core team forum?

 

Regards, Peter

Peter-Simpson, 2020-02-27:
The installer projects are now targeted at 4.7.2 and the Platform Annunciator controls are marked as deprecated in Platform 6.5 with a message pointing at the NuGet package so I'm now closing this issue.

closed 2020-02-27 by Peter-Simpson