ASCOMInitiative/ASCOMPlatform #19: Cannot install on latest Windows 10 Pro

crusty-dave, 2019-11-27:
Microsoft Windows Version 1909 (OS Build 18363.476)

I get the following error:

Error reading from file C:\PROGRA~3\{F4B1C~1\OFFLINE\459C26BD\C4C88ADB\AscomMasterInterfaces.tlb.  Verify that the file exists and that you can access it.

I do not see the folder 459C26BD in the following directory:
C:\ProgramData\{F4B1CB96-DAB5-4F31-AA59-94ECEEFDE0F9}\OFFLINE

The following doesn't log doesn't provide any additional clues as to why it failed.

```
1:55:51 PM ASCOM Platform 6.4 SP1 v6.4.1.2695
1:55:51 PM Installation date: 11/27/2019
...
1:55:51 PM Installer: C:\ProgramData\miaAB61.tmp\ASCOMPlatform64.exe
1:55:51 PM Engine cache: C:\ProgramData\{F4B1CB96-DAB5-4F31-AA59-94ECEEFDE0F9}\
...
1:55:51 PM Product code: {8961E141-B307-4882-ABAD-77A3E76A40C1}
1:55:51 PM Revision code: {F4B1CB96-DAB5-4F31-AA59-94ECEEFDE0F9}

1:55:51 PM Installer mode: MSI Engine

1:55:51 PM OS: Windows TENFOUR
1:55:51 PM Product Name:  Windows 10 Enterprise
1:55:51 PM Edition:  Enterprise
1:55:51 PM Version:  6.3
1:55:51 PM Build:  18363
1:55:51 PM Build Label:  18362.1.amd64fre.19h1_release.190318-1202
1:55:51 PM Architecture: 64bit

1:55:51 PM Install method: EXECUTABLE
...
1:55:51 PM Win64Mode: TRUE
...
1:55:51 PM FrameworkRoot: C:\Windows\Microsoft.NET\Framework\ 
1:55:51 PM RegAsm32: C:\Windows\Microsoft.NET\Framework\v4.0.30319\regasm.exe
1:55:51 PM RegAsm64: C:\WINDOWS\Microsoft.NET\Framework64\v4.0.30319\regasm.exe
1:55:51 PM NGen32: C:\Windows\Microsoft.NET\Framework\v4.0.30319\ngen.exe
1:55:51 PM NGen64: C:\WINDOWS\Microsoft.NET\Framework64\v4.0.30319\ngen.exe
1:55:51 PM ProgramFiles: C:\Program Files (x86)\
1:55:51 PM CommonFiles: C:\Program Files (x86)\Common Files\
1:55:51 PM Maintenance: FALSE, RecommendedUpgrade: FALSE, NeedsUpgrade: FALSE
1:55:51 PM GAC Assembly Reference Path: SOFTWARE\Wow6432Node\Microsoft\Fusion\References
...
1:55:51 PM DesktopDirAll: C:\Users\Public\Desktop\
1:55:51 PM ASCOMStartMenu: ASCOM Platform 6
1:55:51 PM AllUsers: TRUE
1:55:51 PM Target Directory: C:\Program Files (x86)\ASCOM\Platform 6
...

1:55:51 PM Install cache directory: C:\ProgramData\{F4B1CB96-DAB5-4F31-AA59-94ECEEFDE0F9}

1:55:51 PM Platform Version: 6.4.1

1:55:51 PM CheckMSI31 - MSI library found OK
1:55:51 PM CheckNet4 - Found Client framework version: 4.8.03752
1:55:51 PM CheckNet4 - Framework 4 Client present: TRUE
1:55:51 PM CheckNet4 - Found Full framework version: 4.8.03752
1:55:51 PM CheckNet4 - Full Framework 4 Client present: TRUE
1:55:51 PM CheckVC14(86) - VC 14 library (32 bit) found OK
1:55:51 PM CheckVC14(64) - VC 14 library (64 bit) found OK
1:55:51 PM Mode: Install new version
1:55:54 PM Platform Installation Options - TRUE
1:55:54 PM Install Desktop Diagnostics Shortcut - TRUE
1:55:54 PM Install Profile Explorer Shortcut - TRUE
1:55:54 PM Display User Guide - FALSE
 
1:55:54 PM Straight forward install - removing old Platform 4 or 5 if it exists
1:55:54 PM Platform 4.1 - 
1:55:54 PM Platform 5A - 
1:55:54 PM Platform 5B - 
1:55:54 PM First time install, preparing ASCOM Profile Store
1:55:54 PM RC: 0 Set ACL on ASCOM Profile Store
 
1:55:54 PM Installing 32bit Registry entries
1:55:54 PM Creating shortcuts in native mode
1:55:54 PM DesktopFolder: C:\Users\Public\Desktop\
1:55:54 PM ShortcutFolder: C:\ProgramData\Microsoft\Windows\Start Menu\Programs\ASCOM Platform 6

2:06:20 PM Apply Install - Success:  ERROR, LastError: Fatal error during installation

2:06:27 PM UserGuide: User guide display not requested

```

The odd thing is that it installed OK on a Lenovo laptop over a year ago, either the current download or the older download yields the same error on this desktop.

crusty-dave, 2019-11-27:
Apparently the solution was to enable .NET Framework 3.5, I found the following:
https://ascomtalk.groups.io/g/Help/topic/28276666#40044

After that, it installed correctly.

closed 2019-11-27 by crusty-dave

Peter-Simpson, 2019-11-28:
Hi Dave,

 

I’m glad you found the solution.

 

Regards, Peter

 

https://ascomtalk.groups.io/g/Help/topic/28276666#40044