ASCOMInitiative/ASCOMPlatform #22: ASCOM Platform Installer Error

astroman133, 2019-12-12:
An ASCOM user reported that the Platform installer failed to detect and warn that .NET Framework 3.5.1 was not activated on his Win 10 system. According to the current Platform Release Notes, the installation should have provided a popup warning and paused to allow activation of the version 3.5.1 before proceeding on Windows 10. Per the user he received no warning popup but activating v3.5.1 allowed a subsequent installation to complete without error.

Peter-Simpson, 2019-12-13:
This related note was posted by Rick on the Developer forum, so I'm posting it here for reference.

Peter,
I used the Control Panel to remove/deactivate .NET 3.5.1. Then I ran the ASCOM 6.4 SP1 installer and quickly got the "missing 3.5 prerequisite" dialog. I went back to the Control Panel and re-selected the 3.5.1 framework to be active. This used Windows Update to download and install the necessary files. I thought that this is what the OP was reporting.
I thought that he missed the missing prerequisite dialog but he said that he saw no such dialog. It seems as though Framework v3.5.1 was on his PC but was broken in some way that was repaired when he chose to install/activate it.
It seemed that perhaps a more comprehensive check for the required framework version might be beneficial. Perhaps it may be enough to just keep this in mind in case there are more instances of this installation issue.
Thanks!

Peter-Simpson, 2020-01-27:
I've been unable to reproduce this issue where .NET 3.5 is reported as being installed but doesn't actually work. I always see the expected installer messaging whenever .NET 3.5 is flagged as "not installed".

I suspect that the OP's .NET 3.5 installation was corrupt i.e. it was flagged as enabled but was not in fact usable. I don't think that it is warranted to add "proof of function" testing for all of the myriad operating system components and features that are used by the installer. If the OS says that a component is installed I think the installer should be able to rely on this.

This seems to be a one-off incident so I am going to close this issue, it can always be re-opened if similar incidents come to light in future.

closed 2020-01-27 by Peter-Simpson