ASCOMInitiative/ASCOMPlatform #24: Possible bug in ASCOM.Astrometry.AstroUtils.AstroUtils.EventTimes method

emifdez, 2019-12-14:
I found a bug in ASCOM.Astrometry.AstroUtils.AstroUtils.EventTimes method.
It doesn't return sunrise time events for latitude lat = 26.695512 and longitude lon = 100.03105 for the date of December 14, 2019 .
I could observe that changed 1% latitude works correctly. It works if you change the time zone.
I think it fails if the time is too close or is equal to 0:00.
Attached code example:
```
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using ASCOM;

namespace testAscomSunsetRise
{
    class Program
    {
        static void Main(string[] args)
        {
            Console.WriteLine("Hello World!");
            ASCOM.Astrometry.Transform.Transform apTransform = new ASCOM.Astrometry.Transform.Transform();
            ASCOM.Astrometry.NOVAS.NOVAS31 apNOVAS = new ASCOM.Astrometry.NOVAS.NOVAS31();
            ASCOM.Astrometry.AstroUtils.AstroUtils apUtil = new ASCOM.Astrometry.AstroUtils.AstroUtils();
            System.Collections.ArrayList astroNight = new System.Collections.ArrayList();
            System.Collections.ArrayList civilNight = new System.Collections.ArrayList();
            System.Collections.ArrayList sunRiseSet = new System.Collections.ArrayList();
            DateTime lastUpdate = DateTime.UtcNow;
            double lat = 26.695512;
            double lon = 100.03105;
            civilNight = apUtil.EventTimes(ASCOM.Astrometry.EventType.CivilTwilight, lastUpdate.Day, lastUpdate.Month, lastUpdate.Year, lat, lon, 0);
            astroNight = apUtil.EventTimes(ASCOM.Astrometry.EventType.AstronomicalTwilight, lastUpdate.Day, lastUpdate.Month, lastUpdate.Year, lat, lon, 0);
            sunRiseSet = apUtil.EventTimes(ASCOM.Astrometry.EventType.SunRiseSunset, lastUpdate.Day, lastUpdate.Month, lastUpdate.Year, lat, lon, 0);
            Console.WriteLine("Latitude: " + lat.ToString());
            Console.WriteLine("Longitude:" + lon.ToString());
            Console.WriteLine("Time" + lastUpdate.ToString());

            Console.WriteLine("sunRiseSet cont :" + sunRiseSet.Count.ToString());
            foreach (var e in sunRiseSet) Console.WriteLine("Array sunRiseSet:" + e.ToString());
            Console.WriteLine("civilNignht cont :" + civilNight.Count.ToString());
            foreach (var e in astroNight) Console.WriteLine("Array astro:" + e.ToString());

            Console.WriteLine("astroNight cont :" + astroNight.Count.ToString());

            DateTime objDateWoHour = lastUpdate.Date;
        }
    }
}

The output of the example with a bug and changing the latitude (1%).

C:\Users\Isilda\source\repos\testAscomSunsetRise\testAscomSunsetRise\bin\x86\Deb
ug>testAscomSunsetRise.exe

> 
> Hello World!
> Latitude: 26,695512
> Longitude:100,03105
> Time14/12/2019 18:07:22
> sunRiseSet cont :4
> Array sunRiseSet:True
> Array sunRiseSet:0
> Array sunRiseSet:1
> Array sunRiseSet:10,4769542132567
> civilNignht cont :5
> Array astro:True
> Array astro:1
> Array astro:1
> Array astro:22,6261952111538
> Array astro:11,8566759109858
> astroNight cont :5

> C:\Users\Isilda\source\repos\testAscomSunsetRise\testAscomSunsetRise\bin\x86\Deb
> ug>testAscomSunsetRise.exe
> Hello World!
> Latitude: 26,96246712
> Longitude:100,03105
> Time14/12/2019 18:14:55
> sunRiseSet cont :5
> Array sunRiseSet:False
> Array sunRiseSet:1
> Array sunRiseSet:1
> Array sunRiseSet:0,00456997991677
> Array sunRiseSet:10,4673235217848
> civilNignht cont :5
> Array astro:True
> Array astro:1
> Array astro:1
> Array astro:22,632481231502
> Array astro:11,8504289094108
> astroNight cont :5

emifdez, 2019-12-15:
The method works correctly, it was my mistake, I didn't think there was a day without sunrise, but it depends a lot on when the day starts.

> testAscomSunsetRise.exe
> Hello World!
> Latitude: 26,695512
> Longitude:100,03105
> Time:10/12/2019 11:07:55
> sunRiseSet cont :5
> Array sunRiseSet:True
> Array sunRiseSet:1
> Array sunRiseSet:1
> Array sunRiseSet:23,9630359486981
> Array sunRiseSet:10,4581370341065
> Latitude: 26,695512
> Longitude:100,03105
> Time:11/12/2019 11:07:55
> sunRiseSet cont :5
> Array sunRiseSet:True
> Array sunRiseSet:1
> Array sunRiseSet:1
> Array sunRiseSet:23,9739061145482
> Array sunRiseSet:10,4622385314266
> Latitude: 26,695512
> Longitude:100,03105
> Time:12/12/2019 11:07:55
> sunRiseSet cont :5
> Array sunRiseSet:True
> Array sunRiseSet:1
> Array sunRiseSet:1
> Array sunRiseSet:23,9845795058531
> Array sunRiseSet:10,4667449289712
> Latitude: 26,695512
> Longitude:100,03105
> Time:13/12/2019 11:07:55
> sunRiseSet cont :5
> Array sunRiseSet:True
> Array sunRiseSet:1
> Array sunRiseSet:1
> Array sunRiseSet:23,9950463094445
> Array sunRiseSet:10,4716517522506
> Latitude: 26,695512
> Longitude:100,03105
> Time:14/12/2019 11:07:55
> sunRiseSet cont :4
> Array sunRiseSet:True
> Array sunRiseSet:0
> Array sunRiseSet:1
> Array sunRiseSet:10,4769542132567
> Latitude: 26,695512
> Longitude:100,03105
> Time:15/12/2019 11:07:55
> sunRiseSet cont :5
> Array sunRiseSet:False
> Array sunRiseSet:1
> Array sunRiseSet:1
> Array sunRiseSet:0,00526805624205551
> Array sunRiseSet:10,4826471728335

Peter-Simpson, 2019-12-15:
Hi, emifdez,

Yes, we live in a strange world, where some latitudes have days without a sunrise or sunset. I don't know if you saw this example in the ASCOM Developer help : [Almanac Example](https://ascom-standards.org/Help/Developer/html/1c04e43b-7893-477c-be60-7eb8a14f2b85.htm).

The example output for 10th March shows the phenomenon you have run into where that day has no sun rise.

I'm going to close this because it's not an ASCOM bug but a "feature" of our world!

Regards, Peter

closed 2019-12-15 by Peter-Simpson

emifdez, 2019-12-15:
Hi Peter-Simpson,
Thank you very much for the answer and for the example.
I think we reopen the issue. I just executed almanac example, I think there is an error the method.
It is possible that where the sunrise event does not occur (you can see it for the Lijiang observatory in time zone UT, TZ=0, Year = 2020, Latitude = 26.695512, Longitude = 100.03105, date = Dec, 13th 2020). That is correct (when sunrise is late and goes from 23:59 to 00:00 (two days later)), but there must be another day where there are two sunrises. These two sunrises would occur on February 10, 2020, instead just write a single time (first sunrise would be around 00:00 and the other around 23:59).
I edited the next lines:
`        ' Set the year and geographical location for this run
        Const Year As Integer = 2020
        Const Latitude As Double = 26.695512 ' Positive to the North
        Const Longitude As Double = 100.03105 ' Positive to the East
        Const TimeZone As Double = 0' Positive to the East`

And I set the EventType
`     ' Set the event for which the Almanac is required: Sunrise/set, Moonrise/set, Civil, Nautical, Amateur Astronomical or Astronomical twilight
        ' or planet rise/set
        Const TypeOfEvent As EventType = EventType.SunRiseSunset`
I attached the log text at the bottom.
I programing a test (similar to the first but I have executed from Feb 8th to Feb 15th of 2020).
The timespan between sunrises on 9th and 10th is more than 47 hours, That is not possible at Lijiang Observatory. If the time zone es 8, the timespan between adjacent ssunrises is normal (about 24hours).

```
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using ASCOM;

namespace testAscomSunsetRise
{
    class Program
    {
        static void Main(string[] args)
        {
            Console.WriteLine("Hello World!");
            ASCOM.Astrometry.AstroUtils.AstroUtils apUtil = new ASCOM.Astrometry.AstroUtils.AstroUtils();
            System.Collections.ArrayList sunRiseSet = new System.Collections.ArrayList();
            DateTime lastUpdate = new DateTime(2020, 2, 8, 0, 0, 0, DateTimeKind.Utc);
            double lat = 26.695512;
            double lon = 100.03105;
            int Nrises = 0, Nsets = 0;
            DateTime lastSunRise = lastUpdate;
            DateTime lastSunSet = lastUpdate;
            Console.WriteLine("Latitude: " + lat.ToString());
            Console.WriteLine("Longitude:" + lon.ToString());
            for (int i = 0; i < 8; i++)
            {
                Console.WriteLine("Day "+ i.ToString()+" Time:" + lastUpdate.ToString());
                sunRiseSet = apUtil.EventTimes(ASCOM.Astrometry.EventType.SunRiseSunset, lastUpdate.Day, lastUpdate.Month, lastUpdate.Year, lat, lon, 0);
                Console.WriteLine("sunRiseSet cont :" + sunRiseSet.Count.ToString());
                int h = 0;
                foreach (var e in sunRiseSet) {

                    Console.WriteLine("  Array sunRiseSet[" + h.ToString()+ "]= "+e.ToString());
                    h++;
                }
                Nrises = (int)sunRiseSet[1];
                Nsets = (int)sunRiseSet[2];
                Console.Write("Daylight: " + sunRiseSet[0].ToString());
                Console.Write(". Number of sunrises: " + Nrises);
                Console.WriteLine(". Number of sunsets: " + Nsets);

                for (int j = 0; j < Nrises; j++) {
                    //SunRise
                    Console.WriteLine("Sunrise no " + j.ToString() + ": "+ ((double)sunRiseSet[3 + j]).ToString()+"   ");

                }
                for (int j = 0; j < Nsets; j++)
                {
                    //sunSet 
                    Console.WriteLine("SunSet no " + j.ToString() + ": "+ ((double)sunRiseSet[3+Nrises + j]).ToString()+"   ");
                }
[... 155 lines cut ...]
18:18:23.830 Almanac                   12  0810 1846  0759 1910  0732 1927  0659 1943  0634 1959  0626 2014  0635 2016  0650 1959  0704 1928  0718 1854  0737 1831  0759 1828  
18:18:23.859 Almanac                   13  0810 1847  0758 1910  0731 1928  0658 1943  0634 1959  0626 2014  0635 2016  0651 1958  0705 1926  0718 1853  0738 1830  0800 1829  
18:18:23.888 Almanac                   14  0810 1847  0757 1911  0730 1928  0657 1944  0633 2000  0626 2014  0636 2016  0651 1957  0705 1925  0719 1852  0739 1830  0800 1829  
18:18:23.917 Almanac                   15  0810 1848  0757 1912  0729 1929  0656 1944  0632 2000  0626 2015  0636 2015  0652 1957  0705 1924  0719 1852  0739 1829  0801 1829  
18:18:23.946 Almanac                   16  0810 1849  0756 1912  0728 1929  0655 1945  0632 2001  0626 2015  0637 2015  0652 1956  0706 1923  0720 1851  0740 1829  0801 1830  
18:18:23.974 Almanac                   17  0810 1850  0755 1913  0727 1930  0654 1945  0631 2001  0626 2015  0637 2015  0653 1955  0706 1922  0720 1850  0741 1829  0802 1830  
18:18:24.003 Almanac                   18  0810 1851  0754 1914  0726 1930  0653 1946  0631 2002  0626 2016  0637 2014  0653 1954  0707 1921  0721 1849  0742 1828  0803 1830  
18:18:24.031 Almanac                   19  0809 1851  0754 1914  0725 1931  0652 1946  0631 2002  0627 2016  0638 2014  0654 1953  0707 1920  0722 1848  0742 1828  0803 1831  
18:18:24.061 Almanac                   20  0809 1852  0753 1915  0724 1931  0651 1947  0630 2003  0627 2016  0638 2014  0654 1952  0708 1919  0722 1847  0743 1828  0804 1831  
[... cut ...]
18:18:24.782 Almanac                   31  0806 1901             0712 1937             0627 2009             0644 2008  0659 1941             0729 1838             0808 1838  
18:18:24.782                           
18:18:24.782                               **** Always risen or above the requested level                    ---- Always set or below the requested level
18:18:24.782                                    Spaces indicate no event                                          Multiple events in a day are shown as multiple day lines

```