ASCOMInitiative/ASCOMPlatform #28: 6.5 RC1 Chooser unable to delete unwanted remote clients

NameOfTheDragon, 2020-02-27:
In the following screen snipping, notice "ASCOM Remote Client 1" in the drop-down list.

[image]

I wanted to delete this remote client, but the delete screen is empty:
[image]

Peter-Simpson, 2020-02-27:
I did think about this at one point but triaged it out when I ran out of time. 

I've now added the capability to list ASCOM Remote drivers as well as Alpaca Dynamic drivers to the client manager. This is the only change from RC1, does it work for you?

Installers:
[ASCOM Platform](https://download.ascom-standards.org/beta/ASCOMPlatform65.exe)
[Developer Tools](https://download.ascom-standards.org/beta/ASCOMPlatform65Developer.exe)
[Remove ASCOM](https://download.ascom-standards.org/beta/RemoveASCOM.zip)

Peter-Simpson, 2020-07-04:
This change will be in the forthcoming Platform 6.5 so I'm closing this issue.

closed 2020-07-04 by Peter-Simpson