ASCOMInitiative/ASCOMPlatform #29: AstroUtils constructor  in 6.5 RC1 throw NullReferenceException

wsguest, 2020-03-17:
Hi, Peter
I create an ASCOM 6.5 RC1 driver demo using project template(c#) in Visual Studio Community 2019 , without any addtional code, build it and test it with a console application , which is also generated from the template, the driver throw an exception on creating AstroUtils object line:

`astroUtilities = new AstroUtils();`
    
Data:
    System.Collections.ListDictionaryInternal

StackTrace：
    ASCOM.Utilities.RegistryAccess.EnumProfile(String p_SubKeyName) 
   C:\ASCOM Build\Export\ASCOM.Utilities\ASCOM.Utilities\RegistryAccess.vb : line 325

TargetSite : 
    {System.Collections.Generic.SortedList`2[System.String,System.String] EnumProfile(System.String)}

Is that a bug? or any other settings are not correct?

thanks!

PS: I had ever installed ASCOM Platform 6.4 SP1 before on my Win10 laptop.

Peter-Simpson, 2020-03-17:
Hi,

I can't reproduce this, please can you zip up your entire solution and post it here so that i can investigate further?

Thanks, Peter

wsguest, 2020-03-17:
OK, please download attchment file. Thanks a lot.
[FocuserTest.zip](https://github.com/ASCOMInitiative/ASCOMPlatform/files/4343494/FocuserTest.zip)

wsguest, 2020-03-17:
I will try to reinstall ASCOM 6.5 after old versions are completely cleared.

wsguest, 2020-03-17:
> I will try to reinstall ASCOM 6.5 after old versions are completely cleared.

No lucky. I confirmed the ASCOM.Utilities.dll version, it is 6.5.0.2980.

Peter-Simpson, 2020-03-17:
Hi,

Your code works fine here too so I'm wondering whether there is a permissions issue on the ASCOM Profile store registry on your PC.

Please can you create a diagnostics trace, zip it up and post it here as described in this article: https://ascomtalk.groups.io/g/Help/wiki/How-to-create-a-Diagnostics-log 

Thanks, Peter

wsguest, 2020-03-18:
Hi, Peter
That's strange. I run my application in an administration group. permissions should not be the reason.
Diagnostic logs and snapshots:  [ASCOM Logs.zip](https://github.com/ASCOMInitiative/ASCOMPlatform/files/4346552/ASCOM.Logs.zip)

wsguest, 2020-03-18:
When start Conform.exe,  it throws System.MissingMethodException on Conform.FrmConformMain.ConformMainForm_Load :
“Microsoft.Win32.RegistryKey ASCOM.Utilities.RegistryAccess.OpenSubKey(Microsoft.Win32.RegistryKey, System.String, Boolean, RegWow64Options)”

This is probably caused by the same thing about RegistryAccess class.

Peter-Simpson, 2020-03-18:
The Diagnostics log looks fine, but Diagnostics does run with administrator permissions and you said earlier that your application runs OK when it has administrator permissions.

I have never come across this situation before and it suggests that there is something unusual in the security environment on your PC that limits what a normal user can do even through the correct permissions exist on the ASCOM Profile store registry key.

Was Platform 6.4 installed and working OK before you installed Platform 6.5RC1 or is this a Platform Installation starting with Platform 6.5RC1?

Do your registry permissions look like I showed in this post when not running with administrator permissions? https://ascomtalk.groups.io/g/Help/message/41917 

I've also described how to manually add the correct permissions here in case it helps: https://ascomtalk.groups.io/g/Help/message/41910 

Regards, Peter

wsguest, 2020-03-18:
Hi, Peter
> Was Platform 6.4 installed and working OK before you installed Platform 6.5RC1 or is this a Platform Installation starting with Platform 6.5RC1?

I'm not sure if the previous version is OK, and will not confirm that.
(obsessive-compulsive disorder to latest version *_*)

>Do your registry permissions look like I showed in...

Registry permissions are set correctly.

Now, it looks like a problem of my laptop, I will try it on a new computer later.
Once I find the solution，I will post it here.

Many many thanks for your help.

Best regards.

wsguest

Peter-Simpson, 2020-03-18:
Good luck, I hope you find a solution!

Best wishes, Peter

wsguest, 2020-03-19:
I attach souce code to debug, and trace the call stack:

-  	ASCOM.Utilities.RegistryAccess.EnumProfile(String p_SubKeyName) 
-  	ASCOM.Astrometry.EarthRotationParameters.RefreshState()
-  	ASCOM.Astrometry.EarthRotationParameters.New(TraceLogger SuppliedTraceLogger) 
- 	ASCOM.Astrometry.NOVAS.NOVAS31.New() 
-  	ASCOM.Astrometry.AstroUtils.AstroUtils.InitialiseAstroUtils() 
-  	ASCOM.Astrometry.AstroUtils.AstroUtils.New() 

AstroUtils ctor calls method EnumProfile of RegistryAccess class in RegistryAccess.vb,
the 7th line throws NullReferenceException for OpenSubKey failed and return nothing.
`7: Values = ProfileRegKey.OpenSubKey(CleanSubKey(p_SubKeyName)).GetValueNames`
The value of parameter p_SubKeyName is "Astrometry\Latest Leap Second Data", defined in GloabalItems.vb, which can not be found in Registry of my laptop. 
### Solution:
Add subkeys (registry items) to ProfileRegKey HKLM\SOFTWARE\WOW6432Node\ASCOM\Astrometry\
subkeys(items):
Latest Leap Second Data
Latest Delta UT1 Data

regards.

Peter-Simpson, 2020-03-19:
Thanks very much, I'm glad you have a work round but I still don't understand what is happening on your laptop. The EarthRotationParameters.Refresh method is written in VB and contains this code:

```
Try
    ProfileLeapSecondsValueStrings = profile.EnumProfile(AUTOMATIC_UPDATE_LEAP_SECOND_HISTORY_SUBKEY_NAME)
Catch ex As NullReferenceException ' Key does not exist so supply an empty sorted list
    LogDebugMessage("RefreshState", String.Format("Profile key does not exist - there are no downloaded leap second values"))
    DownloadedLeapSecondValues = New SortedList(Of Double, Double)
End Try
LogDebugMessage("RefreshState", String.Format("Found {0} leap second values in the Profile", ProfileLeapSecondsValueStrings.Count))
```

The NullReferenceException is expected and will be caught and handled rather than causing AstroUtils to fail. Are you sure that this is the root of the issue?

Regards, Peter

wsguest, 2020-03-19:
Hi, Peter
I reproduced that on a new computer, by installing ASCOM Platform 6.4SP1, uninstalling it, removing all components by your removal tool, and then installing 6.5 RC1.

I also noticed that try catch block in EarthRotationParameters.Refresh. There is another one in callee EnumProfile method.
Maybe, The root of the issue is accessing property of a nothing(null) object throws a uncatchable NullReferenceException, see below code.(VS 2019, .net Framework 4.8)

```
    Sub RefreshState()
        Try
            EnumProfile("Hello Peter.") ' everything is OK
            EnumProfile(Nothing) ' throws NullReferenceException internally and cannot be caught
        Catch ex As NullReferenceException
            Console.WriteLine(ex.Message)
        End Try
    End Sub

    Function EnumProfile(p As String)
        Try
            Console.WriteLine(p)
            EnumProfile = p.Length 'p.Length throws NullReferenceException
            'Values = ProfileRegKey.OpenSubKey(CleanSubKey(p_SubKeyName)).GetValueNames
        Finally
            Console.WriteLine("bla bla.")
        End Try
    End Function
```
Best regards.

Peter-Simpson, 2020-03-19:
Hi,

OK, thanks for letting me know the cause, I'll close the issue.

Regards, Peter

closed 2020-03-19 by Peter-Simpson