ASCOMInitiative/ASCOMPlatform #30: Enhance Telescope Simulator for .NET

astroman133, 2020-03-20:
The DestinationSideOfPier method is apparently optional because of comments that indicate that it could throw a MethodNotImplemented exception. However, there is no way to simulate or test this behavior except through a driver that exhibits this behavior. It would be helpful for the simulator to be able to optionally throw MethodNotImplemented exception as a result of calls to DestinationSideOfPier to allow application developers to have confidence that they are handling this type of "weak" driver correctly. One of these drivers is the Software Bisque 2X Telescope Driver.

Thanks!

Peter-Simpson, 2020-07-04:
I've added a new DestinationSideofPier enable / disable checkbox to the Telescope Simulator setup as requested. I've also fixed the SideofPier property that I found was not respecting it's setup enable / disable checkbox setting. These changes will be in the Platform 6.5 release.

astroman133, 2020-07-04:
Thanks, Peter. I closed the issue.

closed 2020-07-04 by astroman133