ASCOMInitiative/ASCOMPlatform #31: Observing Condition Hub does not call TimeSinceLastUpdate of underling driver

elendil-software, 2020-04-11:
Hi,

I noticed that the Observing Condition Hub does not call TimeSinceLastUpdate of the underling driver(s). The result is always 0 (or close to zero).

I think the problem is not the drivers, because :

- when the drivers are called directly the values are correct
- the behaviour is the same with several drivers.

I found that at this line : https://github.com/ASCOMInitiative/ASCOMPlatform/blob/cafc66c4b15be4819b48d2e68c0427aa2348c6c5/Drivers%20and%20Simulators/ObservingConditions%20Hub/OCH%20Server/Hub.cs#L784

the condition is never true, so, these lines are executed
https://github.com/ASCOMInitiative/ASCOMPlatform/blob/cafc66c4b15be4819b48d2e68c0427aa2348c6c5/Drivers%20and%20Simulators/ObservingConditions%20Hub/OCH%20Server/Hub.cs#L804-L809

Before investigate﻿, I would like to know if this is a known issue.

Regards,

Julien

Peter-Simpson, 2020-04-11:
Hi Julien,

 

Yes, this is a known bug that is fixed in Platform 6.5 release candidate 1: https://github.com/ASCOMInitiative/ASCOMPlatform/releases 

 

Best wishes, Peter

 

https://github.com/ASCOMInitiative/ASCOMPlatform/blob/cafc66c4b15be4819b48d2e68c0427aa2348c6c5/Drivers%20and%20Simulators/ObservingConditions%20Hub/OCH%20Server/Hub.cs#L804-L809

elendil-software, 2020-04-12:
Great, 
Thanks.

Julien

closed 2020-04-12 by elendil-software