ASCOMInitiative/ASCOMPlatform #33: Setting Transform.JulianDateTT to 0 causes validation exception

jwellman80, 2020-04-27:
Documentation for this property states:

> The inital value of this parameter is 0.0, which is a special value that forces Transform to replicate original behaviour by determining the Julian date from the PC's current date and time. If this property is non zero, that particular terrestrial time Julian date is used in preference to the value derrived from the PC's clock.

It seems that setting this value to 0 should be permitted and pass validation as this is a special case within the interface for this property.

Peter-Simpson, 2020-04-27:
Hi Jared,

Thanks for reporting this, good catch! When I added validity range checking I clearly forgot about the 0.0 special value. I've now fixed this and it will be in the next release candidate that I'm going to issue shortly.

Regards, Peter

Peter-Simpson, 2020-07-04:
This change is in the forthcoming Platform 6.5 release so I'm closing the issue.

closed 2020-07-04 by Peter-Simpson