ASCOMInitiative/ASCOMPlatform #35: Possible deadlock in Utilities.Profile

NameOfTheDragon, 2020-05-26:
Regarding 6.5 RC2

Profile Explorer started throwing exceptions after I used it to delete a device profile.
[image]

This started when I was testing a driver which was getting `NullReferenceException` when trying to restore settings through the `SettingsProvider`. This is not supposed to happen because missing settings are created with a default value. However, I found that the settings were present but with empty values. Thinking this to be an odd situation, I opened Profile Explorer and used it to delete the device's profile (top level settings key). This appeared to work.

I re-ran my driver and still got the `NullReferenceException`. I tried to open Profile Explorer amd it threw the exception in the screen image above. It now does this every time.

The exception message requested that I run ASCOM Diagnostics, which also threw an exception and it looks like there may be a deadlock situation. I believe this could be the issue that @DanielVanNoord mentioned in our last online meeting.

[image]

I have attached an export of my ASCOM device profile registry in case that helps, but I suspect the issue is a deadlock in `Utilities.Profile`.

I don't think I have ever observed this issue prior to running the 6.5 RC.

I have had a look into the code (actually after the last meeting) and there doesn't seem to be any obvious explanation for this. It's worth asking @DanielVanNoord - when you saw this problem, were you using the `SettingsProvider` or going direct to the profile?

[HKLM_Software_ASCOM.zip](https://github.com/ASCOMInitiative/ASCOMPlatform/files/4684621/HKLM_Software_ASCOM.zip)

NameOfTheDragon, 2020-05-26:
Follow-up. While I was completing my issue report, I had left my debug session suspended at the `NullReferenceException`.
[image]

The stack trace looks like this
[image]

I had a look at the `SettingsProvider` code and I think I see a possible failure mechanism. Here's the relevant code in the `SettingsProvider`:

```lang=cs
                try
                {
                    string value = ascomProfile.GetValue(deviceId, item.Name, null, String.Empty);
                    if (String.IsNullOrEmpty(value))
                    {
                        spv.SerializedValue = item.DefaultValue; //[ASCOM-256]
                        Diagnostics.TraceVerbose("Defaulted/empty ASCOM Profile DeviceID={0}, Key={1}, Value={2}",
                                                 deviceId, item.Name, item.DefaultValue.ToString());
                    }
                    else
                    {
                        spv.SerializedValue = value;
                        Diagnostics.TraceVerbose("Retrieved ASCOM Profile DeviceID={0}, Key={1}, Value={2}", deviceId,
                                                 item.Name, value);
                    }
                }
                catch
                {
                    spv.SerializedValue = spv.Property.DefaultValue;
                    Diagnostics.TraceVerbose("Defaulted/missing ASCOM Profile DeviceID={0}, Key={1}, Value={2}",
                                             deviceId, item.Name, spv.PropertyValue);
                }
```

The exception appears to be happening on the first line of my `try {}` block. SettingsProvider expects a null or empty string if there is no saved setting - but instead, it gets an exception (this might be a change in behaviour in `Profile` from previous releases).

This might only be an issue when debugging because the debugger halts on the exception, which is leaving the mutex locked, so any ASCOM apps that launch at that point will fail. However something has changed because I've never seen this issue before when debugging and both @DanielVanNoord  and I have seen it on 6.5 RC builds.

Peter-Simpson, 2020-05-26:
Hi Tim,

 

The Profile component hasn’t changed for a long time but I have found that, on occasion, VS is now stopping at exceptions that it didn’t stop at in past releases. Perhaps it is now ignoring previous configuration or is now honouring exceptions that previously weren’t caught? 

 

These exceptions inside Profile are expected when non-existent values are read and are caught and handled within the profile component. If a debug session interrupt happens in mid operation, as you describe, the mutex will be held until the debug interrupt is released and the code is allowed to complete. All concurrent  Profile calls will freeze trying to get the mutex until the interrupted thread releases the mutex. 

 

I’m not yet convinced there is an issue with the Profile code, I think this could be down to a change in VS behaviour.

 

Regards, Peter

NameOfTheDragon, 2020-05-27:
I agree. Also as pointed out by @astroman133 , Visual Studio seems to be suddenly more alert to exceptions that it previously ignored. Problem solved by ticking this checkbox...
[image]

closed 2020-05-27 by NameOfTheDragon