ASCOMInitiative/ASCOMPlatform #43: Consider documentation change to ICameraV3.SensorType

astroman133, 2021-01-15:
Per Peter Simpson,
All the imaging applications I’m aware of depend on having a valid value for SensorType. 

However the documentation in both ASCOM.DeviceInterface and DriverAccess sections describe the property as Optional.

Also, Conform reports an issue if SensorType raises a PropertyNotImplementedException, as is required when an optional property is not supported.

Consider revising the documentation, perhaps to say that a default value of monochrome should be returned if unavailable or unknown. Another option could be to provide a new sensor type of Unknown.

Peter-Simpson, 2021-09-09:
Conform and the specification are clearly inconsistent and we have three options to move this on:
1. Change Conform so that SensorType is handled as an optional member. 
2. Revise the specification to make SensorType implementation mandatory. 
  a) include a bump in interface version to ICameraV4 because member behaviour is significantly changed.
  b) Without a change to interface version number
3. Continue to live with the inconsistency until a larger interface update is planned.

1. Is straightforward and can be communicated as correcting a Conform bug.
2a. Is hard because of all the overhead of introducing a new interface version
2b. Is easy but leaves us open to accusations of arbitrarily changing the standard without consultation
3. Is easiest but just avoids the issue and we still have to deal with it at a later date.

Perhaps 1. is the best way forward? perhaps something to discuss at Sunday's meeting?

Best wishes, Peter

Peter-Simpson, 2024-08-03:
Closing this issue because I've updated Conform Universal to handle SensorType as optional in line with the specification.

closed 2024-08-03 by Peter-Simpson