ASCOMInitiative/ASCOMPlatform #46: Error when start ASCOM 6.5SP1  diagnostics or connecting some devices

arritrancos, 2021-04-13:
Hello,

When open "ASCOM Diagnostics" on a W7 Pro or connect some devices like a Takahashi focuser, appears a similar message like this

[image]

Error               System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> ASCOM.Utilities.Exceptions.HelperException: Unable to open ephemeris file: C:\Program Files (x86)\Common Files\ASCOM\Astrometry\JPLEPH, RC: 1
   at ASCOM.Astrometry.NOVAS.NOVAS31..ctor() in C:\ASCOM Build\Export\ASCOM.Astrometry\ASCOM.Astrometry\NOVAS31.vb:line 107
   at ASCOM.Astrometry.AstroUtils.AstroUtils.InitialiseAstroUtils() in C:\ASCOM Build\Export\ASCOM.Astrometry\ASCOM.Astrometry\AstroUtils.vb:line 46
   at ASCOM.ActiveFocuser.Focuser..ctor()
   --- End of inner exception stack trace ---
   at System.RuntimeTypeHandle.CreateInstance(RuntimeType type, Boolean publicOnly, Boolean noCheck, Boolean& canBeCached, RuntimeMethodHandleInternal& ctor, Boolean& bNeedSecurityCheck)
   at System.RuntimeType.CreateInstanceSlow(Boolean publicOnly, Boolean skipCheckThis, Boolean fillCache, StackCrawlMark& stackMark)
   at System.Activator.CreateInstance(Type type, Boolean nonPublic)
   at System.Activator.CreateInstance(Type type)
   at ASCOM.DriverConnect.ConnectForm.btnConnect_Click(Object sender, EventArgs e) in C:\ASCOM Build\Export\ASCOM.DriverConnect\ConnectForm.cs:line 245

Regards.
Arritrancos

Peter-Simpson, 2021-06-02:
Hi Arritrancos,

Sorry for the delay in replying, I've only just found your report, is this still an issue for you?

Best wishes, Peter

Peter-Simpson, 2021-09-09:
Closing because there has been no response from originator.

closed 2021-09-09 by Peter-Simpson