ASCOMInitiative/ASCOMPlatform #47: Problem with Telescope Simulator Tracking flag management

astroman133, 2021-07-18:
The simulator has a Profile setting called "Track on / off Support". Presumably the purpose of this setting is to indicate whether the simulated telescope supports changing the Tracking state. However it appears to be not completely implemented.

Turning off "Track on / off Support" does not prevent changing the Tracking flag from either the Simulator U/I or from a connected application. Nor does it throw a PropertyNotImplementedException as specified in the interface specification.

Peter-Simpson, 2021-09-09:
Quite a lot of work to address this as well as time for adequate testing. I propose that we omit this from the SP2 release and include it in the subsequent SP3 release.

Peter-Simpson, 2024-08-03:
This has been fixed at some point and the "Track on / off Support" flag now causes Tracking SET to throw a PropertyNotImplementedException when unchecked.

closed 2024-08-03 by Peter-Simpson