ASCOMInitiative/ASCOMPlatform #48: ASCOM.Astrometry Registry Handle Leak

VPInventions, 2021-08-18:
Some application which use ASCOM.Astrometry experience a leak of handles of a registry key "HKLM\SOFTWARE\WOW6432Node\ASCOM". On my system in a remote observatory the leak rate was about 19000 handles per hours.

After some debugging, it turned out that these handles are opened by RegistryAccess objects allocated by AstroUtils and EarthRotationParameters classes.

Both EarthRotationParameters and AstroUtils classes did not have the Finalize method implemented, so the garbage collector could not properly dispose these objects, including RegistryAccess member objects.

After implementing Finalize methods  (see the attached diff) the leak is not reproducible anymore. I was able to test the fix on my imaging system by simply copying ASCOM.Astrometry.dll with the fix to Windows\assembly\GAC_MSIL and Common Files\ASCOM. There were no noticeable leak after a couple of hours.

So while the problem has been fixed for me, it would be good to have the fix included in official ASCOM package. Other people, who run astrophotography automation software continuously for many days, would benefit from the fix.

Thanks,
Vitali
[AstrometryLeak.txt](https://github.com/ASCOMInitiative/ASCOMPlatform/files/7007821/AstrometryLeak.txt)

Peter-Simpson, 2021-08-19:
Hi Vitali,

Thanks for reporting this issue and providing a fix. I've created new builds, linked below, incorporating your fix, Please can you let me know if the issue is now resolved.

Best wishes, Peter

[Platform](https://download.ascom-standards.org/beta/ASCOMPlatform65SP1.exe)
[Developer Tools](https://download.ascom-standards.org/beta/ASCOMPlatform65SP1Developer.exe)

VPInventions, 2021-08-19:
Hi Peter,

thanks for providing the new build. I've just tested it and can confirm that the leak is gone. 

Best regards,
Vitali

Peter-Simpson, 2021-08-21:
Issue confirmed as fixed.

commit d6e5264f (2021-08-18): Ensure that registry handles are properly released in astroutils and earthrotationparameters. Fixes #48
closed 2021-08-21 by Peter-Simpson