ASCOMInitiative/ASCOMPlatform #50: Correct documentation discrepancies

astroman133, 2021-11-07:
The triple-slash comments generated by the Visual Studio Template for a Switch driver and a local server put comments for several methods into the driver code that are different from the documentation in the Developer Help file. These should be reviewed and corrected, as necessary. Methods needing review are:

MinSwitchValue (help file says "should return 0.0" but code comment says "must return 0.0")
MaxSwitchValue (help file says "should return 1.0" but code comment says "must return 1.0")
SwitchStep (help file says "should return the value 1.0" but code comment says "must return 1.0")
GetSwitchValue (help file says "must not throw a "MethodNotImplementedException" but code comment says "boolean switches must throw a not implemented exception"
SetSwitchValue (help file says "Can throw a MethodNotImplementedException if CanWrite(int16) is False" but code comment says "boolean switches must throw a not implemented exception.")

Also, the generated code for SwitchStep seems to be incorrect. The logging call and throwing the initial exception are commented out in favor of returning a constant 1 from the method.

This might also be an opportunity to improve the documentation of the expected behavior regarding how the SetSwitchValue method should behave in the case of a boolean switch, where the expected values are 0.0 or 1.0, when an intermediate value like 0.5 is provided by a client app. The goal should be to provide guidance so that drivers authors know how the driver needs to behave so that all drivers react the same to this type client error.

Peter-Simpson, 2021-11-09:
Hi Rick,

I've reviewed "should" / "must" use and standardised on "must" for consistency. I've also revised the template comments in line with your suggestions. 

Please can you review and let me know whether the changes address the issues you reported.

Many thanks, Peter

Peter-Simpson, 2021-11-12:
Email from Rick copied here for continuity:
```
Peter,
Thanks for looking at this so quickly!
Comments regarding ASCOM.DeviceInterface/ISwitchV2.cs. I assume that this is the source of what goes into the Developer Help file.

I do not understand the comments for MaxSwitchValue in where it says:
<para>If a two state device cannot report its state, <see cref="MaxSwitchValue"/> must return the value 1.0.</para>
There is similar language in MinSwitchValue and SwitchStep
What I don't understand is why an error returning a switch state should have anything to do with what is returned for MinStateValue, MaxStateValue, or SwitchStep. Shouldn't a 2 state switch always return 0.0 and 1.0 for its min and max and 1.0 for SwitchStep?

Comments regarding the code in DriverTemplates/TemplateSources/src/ASCOM Driver Template (CS)/DeviceSwitch.cs
The comments for SetSwicthValue could be improved by saying that a MethodNotImplementedException should be thrown if CanWrite returns False, rather than "the switch cannot be set" which could be due to some other external error.

Comments regarding the code in DriverTemplates/TemplateSources/src/Ascom LocalServer Template CS/Driver/DeviceSwitch.cs
This file needs to be updated with the changes that were made to the driver template above.

Thanks!
 Rick B 
```

Hi Rick,

### ISwitchV2
The two-state device "can't report its state" text doesn't refer to an error state but to a design state i.e. that the physical device is write-only and has no capability to report its status. In this case device min and max must return 0.0 and 1.0 respectively. the text is to remind people not to throw exceptions in the min and max properties.

### ASCOM Driver Template (CS)
Agreed, I've updated the templates.

### Ascom LocalServer Template CS
Oops, missed that one, thanks, template updated!

Thanks, Peter

Peter-Simpson, 2022-01-11:
Closing because the issue is fixed.

closed 2022-01-11 by Peter-Simpson