ASCOMInitiative/ASCOMPlatform #52: Coordinate Transform uses Apparent if SiteTemperature not set

ghilios, 2022-02-02:
See this code: https://github.com/ASCOMInitiative/ASCOMPlatform/blob/064f498b3ca321f83afb41998961ab03e653f501/ASCOM.Astrometry/ASCOM.Astrometry/Transform.vb#L705

When transforming J2000 coordinates using ASCOM's transform library, topocentric is used only if SiteTemperature is also provided. This parameter shouldn't be required if Refraction is false. In J2000ToTopo, SiteTemperature isn't even used if refraction is disabled.

https://github.com/ASCOMInitiative/ASCOMPlatform/blob/064f498b3ca321f83afb41998961ab03e653f501/ASCOM.Astrometry/ASCOM.Astrometry/Transform.vb#L705

Peter-Simpson, 2022-02-03:
Thanks for reporting the issue George, it shouldn't be like that. I had a look at the code and it's been in place since at least 2014!

Best wishes, Peter

Peter-Simpson, 2022-02-05:
Hi George,

I've checked further into the code today and am unable to reproduce a failure if I do not set the site temperature before calling Transform. In the first release of Transform I failed to initialise SiteTemperature to Double.NaN and so it always appeared to be set to the default double value of 0.0. 

This came to light some while ago but I decided not to change the initialisation to NaN and start throwing exceptions because it would break existing clients that unknowingly depend on having the default value of 0.0 for site temperature.

Is this situation causing an issue in your application? If so, please can you give me some simple example code so that I can investigate further.

Many thanks, Peter

ghilios, 2022-02-06:
Ahh, if there's no issue then that's great! I found this during code inspection while investigating an unrelated issue in a telescope driver.

Thanks!

Peter-Simpson, 2022-02-19:
I'm closing this issue as because it turns out that no fix is required.

Best wishes, Peter

closed 2022-02-19 by Peter-Simpson