ASCOMInitiative/ASCOMPlatform #53: DeviceHub accesses TargetRightAscension/Declination during SyncToCoordinates

ghilios, 2022-02-06:
ASCOM DeviceHub accesses the getters for TargetRightAscension/Declination during SyncToCoordinates (https://github.com/ASCOMInitiative/ASCOMPlatform/blob/ffaedc236a2280590bc9936a6a31c6280cbb106f/Drivers%20and%20Simulators/ASCOM%20Device%20Hub/DeviceHub/DeviceManagers/TelescopeManagerAccess.cs#L822). Per the specification, these getters can fail if they have not yet been set by the client application. SyncToCoordinates shouldn't need these properties to function.

This was reproduced using SharpCap, using a plate solve + sync as the first action after connecting the mount (no slew).

astroman133, 2022-02-11:
I agree that SyncToCoordinates does not need the Target coordinates to function, but that is not why they are being read. TargetRightAscension and TargetDeclination are grouped with other properties that can change from second to second and can be set by applications that are connected directly to the telescope driver. They need to be kept up-to-date and so are periodically scanned. However, any exception thrown by the telescope driver should not be seen by an application unless that application is reading those properties. Also, the behavior implemented in Devkce Hub passes Conform testing.

10:48:39.575 TargetDeclination Read            OK       COM Not Set exception generated on read before write
10:48:39.616 TargetDeclination Write           INFO     Tests moved after the SlewToCoordinates tests so that Conform can check they properly set target coordinates.
10:48:39.621 TargetRightAscension Read         OK       COM Not Set exception generated on read before write
10:48:39.663 TargetRightAscension Write        INFO     Tests moved after the SlewToCoordinates tests so that Conform can check they properly set target coordinates.

10:52:12.658 SlewToCoordinates                 OK       Slewed OK. RA:   18:57:06.87
10:52:12.665 SlewToCoordinates                 OK       Slewed OK. DEC:  01:00:00.00
10:52:12.672 SlewToCoordinates                 OK       The TargetRightAscension property 18:57:06.87 matches the expected RA OK. 
10:52:12.682 SlewToCoordinates                 OK       The TargetDeclination property  01:00:00.00 matches the expected Declination OK. 

At this point I see nothing in Device Hub that violates the ASCOM Telescope specification.

It is the responsibility of the telescope driver to ensure that both the arguments in a call to SyncToCoordinates have been set before returning to the caller. This includes reading of the TargetRA and TargetDec on a different thread while they are being set. in the Sync call.

I will leave this Issue open to gather your comments or clarifications.

closed 2022-03-25 by astroman133

ghilios, 2022-03-25:
@astroman133 sorry I missed your response. I don't think this is a violation of the spec, but as it stands right now using SyncToCoordinates with the device hub will fail until the first slew. This exception propagates to the client, and would effect any use where the first action is to take a picture and sync.

Are you suggesting that a client application needs to set TargetRA/Dec before doing SyncToCoordinates? Is this in the spec?

astroman133, 2022-03-26:
I am not able to duplicate what you are seeing with a simple console program that is controlling the ASCOM Telescope Simulator through DeviceHub.

Here is my code:

// This implements a console application that can be used to test an ASCOM driver
//

// This is used to define code in the template that is specific to one class implementation
// unused code can be deleted and this definition removed.

#define Telescope
// remove this to bypass the code that uses the chooser to select the driver
#define UseChooser

using ASCOM.Utilities;

using System;

namespace ASCOM
{
    internal class Program
    {
        private const string deviceId = "ASCOM.DeviceHub.Telescope";

        static void Main( string[] args )
        {

            string id;

            // Uncomment the code that's required
#if UseChooser
            // choose the device
            id = ASCOM.DriverAccess.Telescope.Choose( deviceId );

            if ( string.IsNullOrEmpty( id ) )
            {
                return;
            }

#else
            id = deviceId;
#endif
            // create this device
            ASCOM.DriverAccess.Telescope device = new ASCOM.DriverAccess.Telescope( id );

            Util util = new Util();

            device.Connected = true;

            // now run some tests, adding code to your driver so that the tests will pass.
            // these first tests are common to all drivers.
            Console.WriteLine( "name " + device.Name );
            Console.WriteLine( "description " + device.Description );
            Console.WriteLine( "DriverInfo " + device.DriverInfo );
            Console.WriteLine( "driverVersion " + device.DriverVersion );

            // TODO add more code to test the driver.

            double ra = device.RightAscension;
            double dec = device.Declination;

            Console.WriteLine( $"Pointing before sync at RA = {util.HoursToHMS( ra )} / Dec = {util.DegreesToDMS( dec )}" );

            ra -= 0.016666;
            dec += 0.25;

            try
            {
                Console.WriteLine( "Calling Telescope.SyncToCoordinates()" );
                device.SyncToCoordinates( ra, dec );
                Console.WriteLine( "Back from Telescope.SyncToCoordinates() with no exception." );

                ra = device.RightAscension;
                dec = device.Declination;

                Console.WriteLine( $"Pointing after sync at RA = {util.HoursToHMS( ra )} / Dec = {util.DegreesToDMS( dec )}" );
            }
            catch ( Exception ex )
            {
                Console.WriteLine( "Exception occurred during call to SyncToCoordinates(). Details follow:" );
                Console.WriteLine();
                Console.WriteLine( ex.ToString() );
            }

            device.Connected = false;
            device.Dispose();
            device = null;

            Console.WriteLine();
            Console.WriteLine( "Press Q to quit" );

            ConsoleKeyInfo ki;

            do
            {
                ki = Console.ReadKey();
            } while ( ki.Key != ConsoleKey.Q );
        }
    }
}

And here is the output from the console program:

name DeviceHub -> Simulator
description Device Hub Telescope
DriverInfo DeviceHub telescope driver. Version: 6.5
driverVersion 6.5
Pointing before sync at RA = 03:45:22 / Dec = -19° 35' 40"
Calling Telescope.SyncToCoordinates()
Back from Telescope.SyncToCoordinates() with no exception.
Pointing after sync at RA = 03:44:22 / Dec = -19° 20' 40"

Press Q to quit

Please let me know if you have any questions.

ghilios, 2022-03-26:
I just checked the source code, and the ASCOM Telescope Simulator's "TargetRightAscension" and "TargetDeclination" property getters do not throw exceptions if the setters have not yet been called. The [ITelescopeV3 documentation](https://ascom-standards.org/Help/Platform/html/P_ASCOM_DeviceInterface_ITelescopeV3_TargetRightAscension.htm) specifies it should throw an InvalidOperationException if the property is read before being set for the first time.

Peter-Simpson, 2022-03-26:
@ghilios Both properties do throw the exceptions required by the interface specification, they are generated by the simulator's CheckRange method. The simulator would not pass Conform if it did not throw those exceptions.

Best wishes, Peter

ghilios, 2022-03-26:
I looked at the code some more, and this doesn't reproduce with the Telescope Simulator because SyncToCoordinates sets the Target properties. See: https://github.com/ASCOMInitiative/ASCOMPlatform/blob/ffaedc236a2280590bc9936a6a31c6280cbb106f/Drivers%20and%20Simulators/Telescope%20Simulator%20.NET/Telescope/Driver.cs#L1183

Is this a required side effect of SyncToCoordinates?

astroman133, 2022-03-26:
The behavior is documented in the Platform Developer Help for SyncToCoordinates. It says 👍 Parameters

RightAscension_
Type: System.Double
The corrected right ascension (hours). **Copied to the TargetRightAscension property.**
Declination
Type: System.Double
The corrected declination (degrees, positive North). **Copied to the TargetDeclination property.**

ghilios, 2022-03-26:
Ahh I see it now. Thank you! I'll update my driver to do the same thing then.

astroman133, 2022-03-26:
George, you should be testing your driver with the Conform tool. Conform will not find all the bugs in your code but it does a very good job of testing the driver's compliance with the ASCOM specification.