ASCOMInitiative/ASCOMPlatform #54: Alpaca discovery fails when ICMP Destination Unreachable reply is received

linuxkidd, 2022-04-10:
**Issue Description:**
* When ASCOM Alpaca discovery is initiated, the broadcast packet goes out over all network interfaces.
* If the interface does not permit broadcast traffic ( Wiregaurd, TailScale ), an 'ICMP Destination Unreachable' is received on the UDP port. ( ICMP Type 3, Code 3 - Port unreachable ).
* This causes the UDPClient code to close the port before any discovery responses can be received and processed.

From [ASCOM.ChooserForm.1153.196840.txt](https://github.com/ASCOMInitiative/ASCOMPlatform/files/8459705/ASCOM.ChooserForm.1153.196840.txt) log file, the following exception is triggered:
```
11:53:19.832 Finder - FinderDiscoveryCallback                           4 - Exception: System.Net.Sockets.SocketException (0x80004005): An existing connection was forcibly closed by the remote host[0D][0A]   at System.Net.Sockets.Socket.EndReceiveFrom(IAsyncResult asyncResult, EndPoint& endPoint)[0D][0A]   at System.Net.Sockets.UdpClient.EndReceive(IAsyncResult asyncResult, IPEndPoint& remoteEP)[0D][0A]   at ASCOM.Utilities.Finder.FinderDiscoveryCallback(IAsyncResult ar) in C:\ASCOM Build\Export\ASCOM.Utilities\ASCOM.Utilities\AlpacaSupport\Finder.vb:line 193
```

Confirmed source is ICMP Destination Unreachable via packet capture:

[image]

**Solution:**
According to this [stackoverflow post](https://stackoverflow.com/questions/38191968/c-sharp-udp-an-existing-connection-was-forcibly-closed-by-the-remote-host), this behavior can be disabled by setting a specific Client.IOControl Code:

```
public const int SIO_UDP_CONNRESET = -1744830452;
var udpClient = new UdpClient(endpoint);
udpClient.Client.IOControl(
    (IOControlCode)SIO_UDP_CONNRESET, 
    new byte[] { 0, 0, 0, 0 }, 
    null
);
```

This appears to be needed around both [Finder.vb line 41](https://github.com/ASCOMInitiative/ASCOMPlatform/blob/7b831d9372e272196e7e8ccf7c8d29bdb4266635/ASCOM.Utilities/ASCOM.Utilities/AlpacaSupport/Finder.vb#L41) and [Finder.cs line 47](https://github.com/ASCOMInitiative/ASCOMPlatform/blob/7b831d9372e272196e7e8ccf7c8d29bdb4266635/ASCOM.Utilities/ASCOM.Utilities.Support/Finder.cs#L47).

I suspect this is impacting a large subset of ASCOM Alpaca users since it seems probable that an ASCOM equipped computer would have a VPN enabled ( especially for remote observatories ).

**Unrelated** - I think there's a typo in [Finder.vb line 45](https://github.com/ASCOMInitiative/ASCOMPlatform/blob/7b831d9372e272196e7e8ccf7c8d29bdb4266635/ASCOM.Utilities/ASCOM.Utilities/AlpacaSupport/Finder.vb#L45), [Finder.vb line 167](https://github.com/ASCOMInitiative/ASCOMPlatform/blob/7b831d9372e272196e7e8ccf7c8d29bdb4266635/ASCOM.Utilities/ASCOM.Utilities/AlpacaSupport/Finder.vb#L167) and [Finder.cs line 52](https://github.com/ASCOMInitiative/ASCOMPlatform/blob/7b831d9372e272196e7e8ccf7c8d29bdb4266635/ASCOM.Utilities/ASCOM.Utilities.Support/Finder.cs#L52).  Specifically, I suspect `ethereal port` is likely meant to be `ephemeral port`.

Peter-Simpson, 2022-04-10:
Hi Michael,

Thanks very much for your comprehensive report and proposed solution. I'm checking with colleagues and will make the changes and issue a test Platform version if the change doesn't induce unforeseen functional impacts.

Best wishes, Peter

linuxkidd, 2022-04-10:
Thanks @Peter-Simpson.

Another thing I noticed - both of the VPN interfaces have a subnet mask of `255.255.255.255`.  Maybe an alternate solution is to not broadcast to such interfaces?  However, this may be but one of multiple possible triggers of the `ICMP Destination Unreachable` response, so I'll defer to your judgement on the best remediation approach.

DanielVanNoord, 2022-04-11:
Hi Michael and Peter,
I looked into this and think we have a more general logic flow issue. Currently in [
FinderDiscoveryCallback](https://github.com/ASCOMInitiative/ASCOMPlatform/blob/7b831d9372e272196e7e8ccf7c8d29bdb4266635/ASCOM.Utilities/ASCOM.Utilities/AlpacaSupport/Finder.vb#L187-L218) we call BeginReceive to resume searching after calling EndReceive to read the response. If EndReceive errors out for any reason (including the ICMP error that Michael found) it will jump straight to the catch and skip restarting the socket. Thus if the error occurs before the real devices finish responding (almost guaranteed because no wire delay to the bad interface) it will cancel any other pending responses.
I think we should move the BeginReceive to a finally block to handle any errors on the EndReceive and read. Micheal, can I send you a simple test program (with and without the fix) for you to try?
Regards,
Daniel

linuxkidd, 2022-04-11:
Hi @DanielVanNoord, sure thing.. Happy to test any code since it's easy for me to reproduce. 

Thanks!

DanielVanNoord, 2022-04-11:
@linuxkidd Thanks, would you rather test building from source (from my discovery samples) or a pre-built binary? I can easily do either.

linuxkidd, 2022-04-11:
Pre-built binary would be best.  I have plenty setup to compile linux code, but would rather not have to establish a full Windows build environment if I can keep from it. :)

DanielVanNoord, 2022-04-11:
@linuxkidd Cool, here are a pair of builds of my discovery testers. The folder labeled Original should error out just like the platform and not find devices and the Fixed one should print the error but continue to discover devices. These just test the UDP part of the discovery, they do not call the HTTP Alpaca APIs.

[FxDiscovery-Original.zip](https://github.com/ASCOMInitiative/ASCOMPlatform/files/8460294/FxDiscovery-Original.zip)
[FxDiscovery-Fixed.zip](https://github.com/ASCOMInitiative/ASCOMPlatform/files/8460287/FxDiscovery-Fixed.zip)

linuxkidd, 2022-04-11:
Original outputs:
```
Searching...
Error message An existing connection was forcibly closed by the remote host
Press Enter to Exit
```

Fixed outputs:
```
Searching...
Got an error from the socket but continued. Error message An existing connection was forcibly closed by the remote host
Hard failure, cannot continue. Error message An existing connection was forcibly closed by the remote host
Press Enter to Exit
```

Do you need any other logging output someplace, or packet captures?

DanielVanNoord, 2022-04-11:
OK, interesting. It looks like Windows kills a client socket completely when it receives a single ICMP reset. Even if you try to resume the socket is dead. This seems like an odd choice.
Two more fixes to try, one creates a socket per interface (which should print the errors and hopefully continue to find stuff) and one that tells the socket to ignore ICMP.

[FxDiscovery-ClientPerInterface.zip](https://github.com/ASCOMInitiative/ASCOMPlatform/files/8460344/FxDiscovery-ClientPerInterface.zip)
[FxDiscovery-ICMPIgnore.zip](https://github.com/ASCOMInitiative/ASCOMPlatform/files/8460345/FxDiscovery-ICMPIgnore.zip)

linuxkidd, 2022-04-11:
Client Per Interface:
```
Searching...
Got an error from the socket but continued. Error message An existing connection was forcibly closed by the remote host
Got an error from the socket but continued. Error message An existing connection was forcibly closed by the remote host
Press Enter to Exit
172.18.122.2:11111 4/10/2022 7:22:03 PM
```

ICMPIgnore
```
Searching...
Press Enter to Exit
172.18.122.2:11111 4/10/2022 7:23:04 PM
```

Much better! :)

DanielVanNoord, 2022-04-11:
Thanks Michael, both results are great. I noticed that the errors came in basically instantly (before the next line of the startup could even print).
I think we should investigate a hybrid approach for the platform / libraries update:
1. Turn off ICMP resets as Michael found in the initial issue. I think we can safely suppress this particular error. This should be blocked in the ASCOMLibrary and Discovery Tests repositories to run on Windows only (I can take care of that in those repositories).
2. Fix the flow in the discovery callback to restart listening (if able) on each interface if a problem does occur. It would not fix this issue but I think it is safer to avoid any future issues.
3. Look at starting a client per interface, that way if an unknown failure occurs or a socket dies on a single interface it will not prevent discovery from running on the other interfaces.

Peter-Simpson, 2022-04-11:
Hi @DanielVanNoord and @linuxkidd,

Thanks very much for assessing the issue and developing and testing the fix. @DanielVanNoord let me know when your changes are in place and I'll update the Library Alpaca clients and Platform Dynamic Clients to match your changes.

Best wishes, Peter

Peter-Simpson, 2022-08-01:
Hi @DanielVanNoord and @linuxkidd,

I've just put a new Platform release candidate here: [ASCOM Platform 6.6 SP1 RC](https://download.ascom-standards.org/beta/ASCOMPlatform66SP1.exe). Please can you let me know whether it fixes the reported issue.

Many thanks, Peter

linuxkidd, 2022-08-03:
Hi @Peter-Simpson , I can confirm that the SP1 RC does indeed allow Alpaca discovery to complete successfully on my system.

Thanks!

Peter-Simpson, 2022-08-03:
Thanks very much for testing the fix Michael!

Best wishes, Peter

commit 0284208b (2022-08-01): Update finder code to match Daniel's revised code in the ASCOM Library. Should fix #54, subject to testing.