ASCOMInitiative/ASCOMPlatform #55: Switch interface does not support operating momentary switches

tonk777, 2022-04-25:
Here is a use case we have:

 _I need to operate momentary action switches via the ASCOM Switch interface that we need to automate at a remote observatory in Chile and operated from the UK. These relay switches are programmed in hardware to remain closed for a set pre-programmed time and then open. These are used to turn on and off 10Micron mounts via their EXT socket and to start up and shutdown NUC computers via the mother board "front panel" header socket.  In particular these momentary action switches are used to shut down hung computers and 10Micron mounts without cycling power to the whole rig.

A small detail here is the regular boot / shutdown occurs when the switch **opens** as long as its not been closed for more than 5 seconds. If the momentary action switch is closed for **>5 seconds** the computer (or mount) is forced to power down at the hardware level.  Hence we have pairs of programmed switches wired in parallel - one switch is programmed as a 2 second momentary action switch and will boot or do a regular shutdown on either NUC computers or 10 Micron mounts. The other switch is programmed to do 8 seconds momentary action to force a hung or unresponsive computer or mount off (it happens !).

So the question is how to interface momentary action switches to the Ascom (Alpaca) Switch interface? The interface supports latched binary switches and multivalued switches but not momentary action swtches

In an ideal world I would use /**setswitch** => true to trigger the hardware momentary switch and expect /**getswitch** to tell the truth and return false when asked - as the switch will have opened again. Note that we want to know the state of the switch - not the hardware the switch controls as there are other ways to start and stop that hardware. The fly in the ointment with this idea is Conform Universal declares this an Issue when /**getswitch** does not return the last value given to /**setswitch**_

Here is a _suggestion_ that could be worked on as the basis for a future extension to the Switch interface if the above use case interests you (I'm doing these as roughed out Alpaca endpoints - switch id arguement is assumed):

- Add a method /**getswitchtype** method returning one of a set of code values interpreted as **BinarySwitch**, **MultiValueSwitch** and **MomentarySwitch**. The later is a new type of switch. This method could support future extensions such as **TimerSwitch** or whatever along with additional methods appropriate for the switch type.

- Add a /**triggerswitch** method, This would initiate the momentary action switch. This method should only be called if  /**getswitchtype** has returned a **MomentarySwitch** code. If it is called for another type of switch it should raise an exception. Other methods such as /**getswitch**, /**getswitchvalue** could also raise the Not Implemented exception for momentary action switches

- Add an optionally implemented /**gettriggerduration** method that indicates the duration the switch closure period. A client can use this to animate a switch graphic to indicate the duration if the momentary nature of the switch. If Not Implemented then the client can animate (or not) the switch graphic in anyway they choose. This method should only be called if  /**getswitchtype** has returned a **MomentarySwitch** code. If it is called for another type of switch it should raise an exception. 

As I said this is a sketch of an idea.

Peter-Simpson, 2022-04-26:
Perhaps best to start with the Conform issue. By default, Conform inserts a delay of 500ms before reading switch values and a delay of 3000ms before writing switch values. Both of these values can be tuned using the Switch tab in Conform's Options / Conformance Options dialogue (image 1 below). Perhaps a smaller read delay will help to ensure that the set value is returned.

Please can you turn on the DisplayMethod Calls and Debug On options (image 2 below), run a test and attach the complete Conform log. This will give me a much better handle on what is happening, particularly on exact timing of calls and responses.

Many thanks, Peter

[image]

[image]

tonk777, 2022-04-26:
Hi Peter,

I'm not reporting a runtime bug but responding to what is believed to be a lack of support for a switch type in the Switch interface. I want to interface momentary action switches but have been told that these are not supported - only latched switches are (ones that report back the last state they were set to, not switches that spontaneously change state at arbitrary times).

I was also told to present the problem here on github and not to continue on ascomtalk. I may have landed this in the wrong place.

Please take a look at the discussion here

https://ascomtalk.groups.io/g/Developer/topic/90672247

Cheers, Tony

[image]<https://user-images.githubusercontent.com/38855929/165259589-695940cc-f696-4c42-8f6f-3853b6d28f81.png>

[image]<https://user-images.githubusercontent.com/38855929/165259345-44331be6-9e8e-4e85-8382-aae4e8ba42a0.png>

Peter-Simpson, 2022-04-27:
OK, thanks Tony, we’ll report back when we have discussed your suggestion. I just thought it may be helpful also to address the point that you made about difficulty in getting your switch to pass Conform, which is intentionally conservative, out of the box, when activating switches to protect equipment.

 

Best wishes, Peter

tonk777, 2022-04-27:
Many thanks Peter,

My driver is now passing conform OK - my issue was just one of (not) synchronising /getswitch and /getswitchvalue - - I just hadn't coupled them correctly at the time in my failed effort to convince the interface to "allow" momentary action switches under conformance testing.

Regarding momentary action switches in the current situation. A client application for my device is able to use these switches if they understand that toggling a regular binary switch from "off" to "on" and also "on" to "off" does exactly the same thing - it triggers the momentary action on each switch "movement"

Regards, Tony

Peter-Simpson, 2022-04-27:
Hi Tony,

 

I had a GM2000HPS for several years, in my garden rather than remote though, and was very pleased with it, so understand your requirement to restart the controller. Have you considered making only one of the off-on / on-off transitions actually trigger the timed relay, so that it only fires once for a complete cycle of the switch from off to on and back to off again? The electronic behaviour would then match the physical behaviour where the reset occurs when the switch re-opens after being momentarily closed, and perhaps could avoid some application complexity.

 

Best wishes, Peter

tonk777, 2022-04-27:
Nice idea. 

However I would eventually hope to avoid such work-rounds  ;). Looking forward to future feedback - even if its "use the work-round". At least it should be documented how to handle this class of switch whatever way its done

Peter-Simpson, 2024-08-03:
Closing this issue because I've added the work-round to our master interface document Switch FAQ: [https://www.ascom-standards.org/newdocs/switch-faq.html](https://www.ascom-standards.org/newdocs/switch-faq.html) (see Question/Answer at end of document).

closed 2024-08-03 by Peter-Simpson