ASCOMInitiative/ASCOMPlatform #58: Determine Bit Depth of Returned ImageArray

mpoelzl, 2022-08-15:
- The ASCOM ICamera interface does not provide a reliable way to tell the app developer the bit depth of the returned image array (i.e. full white could be 4095 or 65535 etc.). MaxADU does not work as it returns the maximum capability, not the actual format.
- In this thread Bob Denning states that all data in the ImageArray should be returned in 16bit data, upscaled if necessary ([https://ascomtalk.groups.io/g/Developer/topic/74883287﻿](url)). However this is not visible in the interface specification.
- As a result ASCOM drivers are developed which do NOT return 16bit data (latest point in case the SvBony ASCOM driver for the SV305 camera).
- App developers need to implement workarounds for specific drivers which runs agains the idea of a universal interface for all cameras.

Possible solutions:

- Clearly state in the interface specification that 16bit is required in the image array to make it clear to ASCOM driver developers that this is the expectation.
- Or add a property to the ASCOM ICamera interface that specifies the format of the returned data.

Peter-Simpson, 2022-08-25:
Hi,

MaxADU tells you the largest bit depth image that the camera can return, so that you know in advance and can put bounds on how you process the image.

We appreciate that if the image is under exposed, the returned data could have a smaller bit depth than MaxADU, but it is the application's responsibility to handle whatever image data the camera delivers. We don't see is as the camera's responsibility to provide image quality metadata, which is what you seem to be requesting here.

We see it as the application's responsibility to determine the actual image bit depth, just as it is the application's responsibility to deal with dead pixels, bright pixels, contrast, image stretching, stacking etc.

This isn't a request that we believe will receive support from camera manufacturers.

Best wishes, Peter

mpoelzl, 2022-08-25:
Thanks for your thoughts but I respectfully disagree :-)!

- I would argue that the bit depth of the returned data is not 'image quality metadata' as suggested but a fundamental property of the dataset and required information to correctly display an image in the application (same as for example image height and width). Similarly ASCOM also provides properties for sensor data and bayer offsets presumably for similar reasons.

You say that it is the applications responsibility to determine the actual bit depth of the returned data. Could you please explain to me how this can be achieved?

Peter-Simpson, 2022-09-06:
Can we be clear on what you mean by bit depth? If I have understood correctly you are using this term to mean:
- The maximum pixel value in the returned image array
- The value can vary from image to image depending on subject brightness and exposure.
- The value may be less than MaxADU

Are these correct?

Best wishes, Peter

mpoelzl, 2022-09-13:
Hi Peter,

Sorry no, are talking about different things. When I refer to 'bit depth' of the image data I mean the total range of values a pixel COULD have i.e. for 12bit data a pixel value can range from 0 to 4095. My main motivation is to be able to correctly dipsplay the image to the user. To be able to do so I need to know this range otherwise (from only looking at the image data) there is no way of knowing if it is a bright 12bit (max is 4095) image or if it is a very dark 16bit image (max is 65535).

MaxADU ('Reports the maximum ADU value the camera can produce') might seem like the answer here but it is not. Depending on camera settings, the image data can have varying bit depths for the same camera so knowing the maximum ADU value a camera can produce does not really give an indication of the actual bit depth of the returned array. For example a 12bit camera can be set to return an 8bit image etc. Some camera manufacturers do actually use MaxADU to give the bit depth of the returned data at any given time which is great but some do not and in any case it seems contrary to the definition.

The biggest problem however is the inconsistency of implementation across popular ASCOM drivers. I would argue this is because the ASCOM framework does not really provide any guidance on a uniform way of doing this. Just to illustrate this, here are some samples of cameras used via the ASCOM driver and they way they return the data array:

- ASI ZWO 120MC-S: 12bit native camera -> Image data returned upscaled to 16bit (MaxADU: 65535)
- SvBony 305C: 12bit native camera - Image data returned at native 12bit (MaxADU: 4095)
- QHY 485: 12 bit native camera -> Image data returned upscaled to 16bit (MaxADU: 255)
- Altair 290C: 12 bit native camera -> Image data returned at native 12bit (MaxADU: 4095)

So there seem to be two conventions, one to upscale the data to 16bit and one to return the data at native bit depth. Either is fine with me but for a random camera there is no way to know beforehand which it will be.

So hence my suggestion to formalise the bit depth of returned data or to include a property similar to MaxADU but with the express definition to return the actual bit depth of the array that will be returned under the current settings at that time.

Kind Regards,

Mike

Peter-Simpson, 2022-11-01:
Hi Mike,

OK, I see your point that upscaling does hide the real bit-depth. However, you state your requirement as: 

> “My main motivation is to be able to correctly display the image to the user.”. 

That is exactly what MaxADU is there to enable. Each pixel’s brightness, expressed as a fraction between 0 and 1.0, is [PixelValue]/[MaxADU]. You can then scale the brightness fractions as you need to display the image.

Of your example cameras it seems to me that the ASI ZWO 120MC-S, SvBony 305C and Altair 290C all provide MaxADU values that enable you to meet your requirement. 

The QHY 485 looks wrong because returning a MaxADU of 255 for a camera operating in 12bit mode is not compliant with the Camera standard; it should return a MaxADU of 4095. On the other hand, if the camera is actually operating in 8bit mode, even though it is 12bit capable, then a MaxADU of 255 would be correct.

If camera drivers do not set MaxADU values that correspond with the maximum brightness that the returned image could have, that is a bug in the driver in my view and should be reported to the manufacturer.

I don't think upscaling is the issue here, you can still meet your requirement regardless of whether or not the image is upscaled, so long as the driver returns a MaxADU value that is correct for the returned image.

Best wishes, Peter

Peter-Simpson, 2024-08-03:
Closing due to no further inactivity.

closed 2024-08-03 by Peter-Simpson