ASCOMInitiative/ASCOMPlatform #59: Camera V3 Simulator Unable To Set Sub Exposure Rate

michealroberts, 2022-09-23:
Currently, utilising Camera V3 Simulator leaves it unable to set the sub exposure rate of the camera.

The ASCOM Alpaca API documentation states: "Only available in Camera Interface Version 3 and later."

But from our integration tests, we seem to not be able to set it. Our code for setting the value is as follows:

```go
/*
	SetSubExposureDuration()

	@returns an error or nil, if nil sets the sub exposure duration in seconds, *only available in Camera Interface Version 3 and later.
	@see https://ascom-standards.org/api/#/Camera%20Specific%20Methods/put_camera__device_number__subexposureduration
*/
func (c *Camera) SetSubExposureDuration(subExposureDuration float64) error {
	c.Alpaca.TransactionId++

	var form map[string]string = map[string]string{
		"SubExposureDuration": fmt.Sprintf("%f", subExposureDuration),
		"ClientID":            fmt.Sprintf("%d", c.Alpaca.ClientId),
		"ClientTransactionID": fmt.Sprintf("%d", c.Alpaca.TransactionId),
	}

	return c.Alpaca.Put("camera", c.DeviceNumber, "subexposureduration", form)
}
```

And for getting the sub exposure duration:

```go
/*
	GetSubExposureDuration()

	@returns the sub exposure duration in seconds, *only available in Camera Interface Version 3 and later.
	@see https://ascom-standards.org/api/#/Camera%20Specific%20Methods/get_camera__device_number__subexposureduration
*/
func (c *Camera) GetSubExposureDuration() (float64, error) {
	return c.Alpaca.GetFloat64Response("camera", c.DeviceNumber, "subexposureduration")
}
```

Yet, our integration test:

```go
func TestNewCameraSetSubExposureDuration(t *testing.T) {
	camera.SetConnected(true)

	camera.SetSubExposureDuration(90)

	var got, err = camera.GetSubExposureDuration()

	if err != nil {
		t.Errorf("got %q", err)
	}

	if got != 90 {
		t.Errorf("got %v, but expected the subframe exposure duration to be set correctly", got)
	}

	if got < 0 && got > 10000 {
		t.Errorf("got %v, but expected the subframe exposure duration to be a realistic value", got)
	}

	if camera.Alpaca.ErrorNumber != 0 {
		t.Errorf("got %q", camera.Alpaca.ErrorMessage)
	}
}
```

Yields the following error:

```
camera_test.go:1370: got 0, but expected the subframe exposure duration to be set correctly
```

So, effectively, the sub-exposure rate seems to be set to 0. Is there anything we need to do to enable this setting in the Camera V3 Simulator?

Peter-Simpson, 2022-09-28:
Hi Michael,

The Alpaca protocol requires that you test the value of the JSON "ErrorNumber" field in the [returned response](https://ascom-standards.org/api/#/Camera%20Specific%20Methods/get_camera__device_number__subexposureduration) in order to determine whether the call was successful in an "ASCOM" sense, as opposed to a communications / networking sense. My guess is that after setting or getting the SubExposure value, you will find that the ErrorNumber field has a value of 0x400, indicating that the called member is not implemented and that there is a supporting message in the "ErrorMessage" field.

By default, both the Platform camera simulator and the Alpaca camera simulator disable SubExposureDuration, and you need to manually enable this capability through the simulator setup GUI. You just need to make sure that the "Has Sub Exposure" checkbox in the "Exposure" section is checked.

I've just tested both simulators and they work as expected when support is enabled.

Hope this helps, 

Best wishes, Peter

Peter-Simpson, 2022-09-28:
Glad that explained what was going on. Good luck with your development!

Best wishes, Peter