ASCOMInitiative/ASCOMPlatform #61: Consider Implementing a slewtocoordinatesasync method for Dome

michealroberts, 2022-09-28:
As the title suggests, it would be nice to have a slew to coordinates asynchronously which matches that of the telescope specific API methods, and alt-az tracking.

Peter-Simpson, 2022-10-10:
Hi Micheal,

Thanks for this suggestion, which, like your issue #60, we also discussed yesterday.

We already have `Dome.SlewToAltitude` and `Dome.SlewToAzimuth` members in the interface and in our view, adding a new member that simply combines the outcomes from these two commands doesn't deliver sufficient value to justify the interface change. 

However, your point about asynchronous operation is well taken and, as part of the next Dome interface, we do intend to require that the `Dome.SlewToAltitude` and `Dome.SlewToAzimuth` methods operate asynchronously. This will enable you to create your own `SlewToAltAzAsync` method that fires off two concurrent tasks to slew to Alt and Az and waits for both to complete asynchronously.

Best wishes, Peter

michealroberts, 2022-10-10:
Hi Peter,

That's great, I appreciate the compromise. 

To keep things consistent with the Telescope API (https://ascom-standards.org/api/#/Telescope%20Specific%20Methods/put_telescope__device_number__slewtoaltazasync) shouldn't we implement both a synchronous call (keeping both `Dome.SlewToAzimuth` and `Dome.SlewToAltitude`) and introduce the async methods as `Dome.SlewToXAsync` (where X is either Azimuth or Altitude)? 

I'd argue for the extra async functions, as we would need both the async call (so to make the request and response cycle close early), and then SSE poll for the { alt, az } position. However, it is also useful to have the sun version for a procedural view of the return, so we don't need to long poll for the sync to target.

Let me know if that makes sense! Essentially, I would argue that there is a strong need for both the sync and async versions of the SlewTo methods for the Dome.

I'd appreciate this to be taken into consideration for the final implementation.

Peter-Simpson, 2022-11-01:
Hi Michael,

I don't know whether you noticed our public release of the ASCOM Library over the week end. Amongst other things this includes both Alpaca and COM clients that have asynchronous methods that return awaitable .NET Tasks for all the long-running interface operations such as slewing telescopes and domes.

These encapsulate the operation initiation and completion variable polling activities in one method. Here are links to the [introduction](https://ascom-standards.org/library/html/d50074a5-fc2e-4a40-a4aa-dec52045953b.htm), [api](https://ascom-standards.org/library/html/T_ASCOM_Common_ClientExtensions.htm) and some basic [use examples](https://ascom-standards.org/library/html/cc7f1bae-15ed-4d69-b238-5d857d570c88.htm).

Hopefully the library will make your development more straightforward.

Best wishes, Peter

michealroberts, 2022-11-01:
Hi Peter,

That's excellent! 🚀 :)

Do you know when these changes will also be reflected in the ASCOM alpaca Simulators / API? Do you need a separate PR for this? I'm happy to contribute ...

Many thanks,

Michael

Peter-Simpson, 2024-08-03:
Closing due to no recent activity.

closed 2024-08-03 by Peter-Simpson