ASCOMInitiative/ASCOMPlatform #62: Consider Implementing an isexposing method for Camera

michealroberts, 2022-10-03:
Currently, to obtain / check that the camera isn't currently exposing an image, I use a combination of the `lastexposureduration` and `lastexposurestarttime`, and combine and apply the maths to see if it is safe to start another exposure.

I propose that this logic can be handled by a `GET` request to `camera/{device_number}/isexposing` route ...

DanielVanNoord, 2022-10-03:
This already exists in the API. See https://ascom-standards.org/Help/Platform/html/P_ASCOM_DeviceInterface_ICameraV3_CameraState.htm and https://ascom-standards.org/api/#/Camera%20Specific%20Methods/get_camera__device_number__camerastate

DanielVanNoord, 2022-10-03:
I would recommend posting development questions in the ASCOM Developers forum first. You can find it here: https://ascomtalk.groups.io/g/Developer/. Often it can be resolved there, otherwise a change proposal can be put together.