ASCOMInitiative/ASCOMPlatform #64: RightAscensionRate on .NET simulator

BobDenny, 2023-02-14:
With Sidereal tracking active (Tracking = True) if you set RightAscensionRate to 30, the TelescopeSimulator.NET shows RA _decreasing_ and much too slowly. Doing the same on the original (Jon Brewster) VB6 Simulator responds as expected, changing the RA by 30 (sidereal) seconds per (sidereal) second. Both RightAscensionRate and DeclinationRate should be validated in the .NET simulator for all three AlignmentModes.

Peter-Simpson, 2023-02-14:
Hi Bob,

Is this the issue that I worked on with Patrick Wallace's help last month? I thought that I had fixed it for all mount alignments in this updated Platform (not yet in a GitHub release): [https://ascomtalk.groups.io/g/Developer/message/6145](https://ascomtalk.groups.io/g/Developer/message/6145) .

Best wishes, Peter

BobDenny, 2023-02-14:
Sorry to report, no. The RA decreases when given a positive RightAscensionRate. I tested with simple equatorial and GEM  on both sides of the meridian, so it is a simple sign reversal somewhere. I think this new simulator solution must be on a branch because I can't see the new one here. Here is a small script  that will show this. to run it

```
> cscript RateOffsetTests.js
Start tracking
Move RA 10 sec per sidereal sec
Now watch the RA on the scope for 6 sec
RA should have increased 1 minute / 60 sec
Freeze RA to sky -> sidereal rate
Now move Dec 10 arcsec per clock sec
Now watch the Dec on the scope for 6 sec
RA should have increased 1 arc minute / 60 sec
Freeze Dec to sky, end of tests
```

Peter-Simpson, 2023-02-15:
Hi Bob,

Thanks for raising the issue and for the test script, it looks like this issue has been there since day 1 of the .NET simulator. I've applied the necessary sign fixes and tested with polar, German polar and AltAz alignment modes, all four SideOfPier quadrants (for German mounts) and northern and southern hemisphere latitudes. All now seems good here.

Please can you let me know if you also think this is fixed?

[https://download.ascom-standards.org/beta/AscomPlatform662beta.3672.exe](https://download.ascom-standards.org/beta/AscomPlatform662beta.3672.exe) 
[https://download.ascom-standards.org/beta/AscomDeveloper662beta.3672.exe ](https://download.ascom-standards.org/beta/AscomDeveloper662beta.3672.exe )

Many thanks, Peter

BobDenny, 2023-02-17:
The most recent .3677 Platform fixes this and additional problems. Closing this one, thank you!

commit c6c75d33 (2023-02-15): Fixes #64.  RA/Dec rate offsets now have the correct sign behaviour: positive rate offsets result in increasing RA and Dec values and vice versa.
Tested with polar, German polar and alt/az mount alignments, all four side of pier quadrants and locations in northern and southern hemispheres.
closed 2023-02-17 by BobDenny