ASCOMInitiative/ASCOMPlatform #66: ASCOM Driver Template missing in VS2022

Bruno-Muller, 2023-07-03:
Hi,

ASCOM Driver Templates are missing in VS2022

Step to reproduce:

Install **Platform AscomPlatform661.3673.exe**
Install **AscomDeveloper661.3673.exe**
In Visual Studio 2022 > Manage Extension > install **ASCOM Driver Project Templates (VS2022)** 

In Visual Studio 2022 > Create a new project, 
"Test Console App" is present
"Test Forms App" is present
"LocalServer Template" is present
**"Driver Template" is missing**

elendil-software, 2023-07-05:
Hi, 

I have the same issue.

Peter-Simpson, 2023-07-05:
Hi Bruno, Julian,

We no longer include the in-process driver templates because local server drivers provide more robust solutions than in-process drivers. A local server driver runs in its own process rather than running within the client application's process as is the case with in-process drivers. 

This means that any really serious driver issues are confined just to the driver and can not bring down the client application, as can happen when drivers run in the client's process. In addition a single local server can provide multiple drivers of the same or different device types and intrinsically acts as a hub that supports multiple clients .

I've copied the short release notes below that were included with the template extension. Unfortunately I'm not aware of a way to install earlier extension versions.

Best wishes, Peter

Local server drivers are the most effective way to deliver drivers because they provide built-in "hub" functionality and isolate the driver from the application by running it in a separate process.

The local server template is now much easier to use than in past templates and in consequence all in-process driver templates have been removed from this release.

Bruno-Muller, 2023-07-06:
Hi Peter,

Thank you.

I think I have been mislead by the documentation file "Introduction to Developing ASCOM Drivers.pdf", it is still referring to the "ASCOM Device Driver CSharp template" on page 2.

In the meantime, I created the project from ASCOM 65SP1 (shame on me).

Peter-Simpson, 2024-08-03:
ASCOM Platform 7 includes a re-written "How to create a driver" help document targeted at the local server template.

Closing because of no recent activity.

closed 2024-08-03 by Peter-Simpson