ASCOMInitiative/ASCOMPlatform #67: InvalidCastException while connecting ASCOM Device hub to SynScanApp

fsignoret, 2023-09-24:
Hello,
I have a new computer, and installed the latest versions of ASCOM Device Hub and SynScanApp ASCOM driver/app.
The ASCOM diagnostics are fine, but it fails in the Device Hub interface:

22:13:04.969: Telescope - Others:       Get Connected flag - False
22:13:04.970: Telescope - Others:       Set Connected flag -> True (connected)
22:13:08.035: Telescope - Others:       Connection took 0 milliseconds.
22:13:08.035: Telescope - Others:       Get Connected flag - True
22:13:09.043: Telescope - Capabilities: Get CanFindHome: False
22:13:09.045: Telescope - Capabilities: Get CanPark: True
22:13:09.047: Telescope - Capabilities: Get CanPulseGuide: True
22:13:09.048: Telescope - Capabilities: Get CanSetDeclinationRate: False
22:13:09.049: Telescope - Capabilities: Get CanSetGuideRates: True
22:13:09.050: Telescope - Capabilities: Get CanSetPark: True
22:13:09.050: Telescope - Capabilities: Get CanSetPierSide: True
22:13:09.051: Telescope - Capabilities: Get CanSetRightAscensionRate: False
22:13:09.052: Telescope - Capabilities: Get CanSetTracking: True
22:13:09.052: Telescope - Capabilities: Get CanSlew: True
22:13:09.052: Telescope - Capabilities: Get CanSlewAltAz: True
22:13:09.053: Telescope - Capabilities: Get CanSlewAltAzAsync: True
22:13:09.053: Telescope - Capabilities: Get CanSlewAsync: True
22:13:09.054: Telescope - Capabilities: Get CanSync: True
22:13:09.054: Telescope - Capabilities: Get CanSyncAltAz: True
22:13:09.055: Telescope - Capabilities: Get CanUnpark: True
22:13:09.055: Telescope - Capabilities: Calling CanMoveAxis( Primary ) - returned True.
22:13:09.056: Telescope - Capabilities: Calling CanMoveAxis( Secondary ) - returned True.
22:13:09.056: Telescope - Capabilities: Calling CanMoveAxis( Tertiary ) - returned False.
22:13:09.057: Telescope - Capabilities: Calling AxisRates( Primary )(done)
22:13:09.057: Telescope - Capabilities: Calling AxisRates( Secondary )(done)
22:13:09.058: Telescope - Capabilities: Calling AxisRates( Tertiary )(done)
22:13:09.058: Telescope - Parameters:   Get AlignmentMode: algAltAz
22:13:09.059: Telescope - Parameters:   Get ApertureArea Exception: Property read  is not implemented in this driver.
22:13:09.060: Telescope - Parameters:   Get ApertureDiameter Exception: Property read  is not implemented in this driver.
22:13:09.060: Telescope - Parameters:   Get Description: SynScan App Driver
22:13:09.061: Telescope - Parameters:   Get DriverInfo: SynScan App Driver 1.4.0
22:13:09.061: Telescope - Parameters:   Get DoesRefraction: False
22:13:09.061: Telescope - Parameters:   Get DriverVersion: 1.4.0
22:13:09.062: Telescope - Parameters:   Get EquatorialSystem: equLocalTopocentric
22:13:09.062: Telescope - Parameters:   Get FocalLength Exception: Property read  is not implemented in this driver.
22:13:09.063: Telescope - Statuses:     Get GuideRateDeclination: 0.00208903730826
22:13:09.063: Telescope - Statuses:     Get GuideRateRightAscension: 0.00208903730826
22:13:09.063: Telescope - Parameters:   Get InterfaceVersion: 3
22:13:09.064: Telescope - Parameters:   Get SiteElevation: 0
22:13:09.064: Telescope - Parameters:   Get SiteLatitude: 43.6233333333
22:13:09.064: Telescope - Parameters:   Get SiteLongitude: 7.04166666667
22:13:09.065: Telescope - Parameters:   Get SlewSettleTime Exception: Property read  is not implemented in this driver.
22:13:09.065: Telescope - Parameters:   Get SupportedActions:  ClearSyncTo
22:13:09.066: Telescope - Others:       Attempting to initialize telescope operation caught an unhandled exception. Details follow:
22:13:09.066: Telescope - Others:       System.InvalidCastException: Impossible d'effectuer au moins un cast d'élément du tableau source dans le type tableau de destination.
   à System.Array.Copy(Array sourceArray, Int32 sourceIndex, Array destinationArray, Int32 destinationIndex, Int32 length, Boolean reliable)
   à System.Collections.ArrayList.ToArray(Type type)
   à ASCOM.DeviceHub.TelescopeParameters..ctor(TelescopeManager mgr) dans C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\Business Object Classes\Telescope Classes\TelescopeParameters.cs:ligne 37
   à ASCOM.DeviceHub.TelescopeManager.ReadInitialTelescopeDataTask() dans C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\DeviceManagers\TelescopeManager.cs:ligne 992
   à ASCOM.DeviceHub.TelescopeManager.<Connect>b__234_0() dans C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\DeviceManagers\TelescopeManager.cs:ligne 220
   à System.Threading.Tasks.Task.InnerInvoke()
   à System.Threading.Tasks.Task.Execute()
22:13:09.066: Telescope - Others:       Set Connected flag -> False (disconnected)
22:13:09.568: Telescope - Others:       Disconnection took 0 milliseconds.
22:13:09.568: Telescope - Others:       Get Connected flag - 
22:13:09.568: Telescope - Others:       Get Connected flag - 

Thanks for your support!
Regards,
Florian

fsignoret, 2023-10-01:
It looks related to the recent changes made in the Skywatcher ASCOM driver (v140) and related App.
There is no issue with version 131, and no "ClearSyncTo" supported action.

DanielVanNoord, 2023-10-03:
Hello Florian,
Looking at the error message it looks like the Skywatcher SynScanApp has added a SupportedAction that is not a string. That is a specification violation that ConformU (https://github.com/ASCOMInitiative/ConformU) should catch. Have you reached out to Skywatcher and let them know that their updated driver does not work correctly?
Regards,
Daniel

Peter-Simpson, 2023-10-03:
And here is the interface specification saying that the response should be an array of strings: https://www.ascom-standards.org/help/developer/html/P_ASCOM_DeviceInterface_ITelescopeV3_SupportedActions.htm

Best wishes, Peter

fsignoret, 2023-10-03:
Hello Daniel
Thanks for the reply.
I tried to reach them by email, but no response so far. I did not find any other way to contact them (no github, no Jira).
Clear skies
Florian

fsignoret, 2023-10-03:
Hello, I got an answer from the syn scan app team.
I will let you know.

BobDenny, 2023-10-04:
For the record here they do have a support page: https://www.skywatcherusa.com/pages/support-center

BobDenny, 2024-02-14:
I am closing this issue now.

closed 2024-02-14 by BobDenny