ASCOMInitiative/ASCOMPlatform #69: ASCOM Alpaca compatibility with existing applications and SetupDialog

astrogene1000, 2023-12-08:
Running a higher level program, e.g. Nina or SGP or even a wscript, once the Alpaca endpoint device is 'Linked/Connected' there is no way to pass through a call made by the higher level program to the Alpaca endpoint.
All drivers seem to block similar to this for Focuser
Github:
ASCOMPlatform/ASCOM.Utilities/AlpacaDynamicClients/Alpaca Client Device Base Classes
/Focuser Base Class.cs

Code lines 213ff
public void SetupDialog()
        {
            TL.LogMessage(clientNumber, "SetupDialog", "Connected: " + clientIsConnected.ToString());
            if (clientIsConnected)
            {
                MessageBox.Show("Simulator is connected, setup parameters cannot be changed, please press OK");
            }
            else
            {

Legacy COM drivers are the ones that make the choice whether SetupDialog can be executed, the decision is not made in the platform code.

Attached is a pic of SGP connected COM to the same device Nina is connected via Alpaca framework when the user click's 'Setup' in the higher level applications.
[image]

And a pic of a wscript calling SetupDialog to the existing Telescope.net simulator, the driver makes the decision to not accept, not the framework.
[image]

The above comes from the driver, not the framework
Github:
From here:
ASCOMPlatform/Drivers and Simulators/Telescope Simulator .NET/Telescope/Driver.cs

 public void SetupDialog()
        {
            if (TelescopeHardware.Connected)
                throw new InvalidOperationException("The hardware is connected, cannot do SetupDialog()");
            try

The above coupled with the erroneous messages by the Alpaca device base classes that the 'Simulator' is connected, the messages do not indicate that a real alpaca endpoint is connected.

:

astrogene1000, 2023-12-13:
Hi Peter,
In legacy COM ASCOM, it is the driver itself that does the 'protection'.
In Alpaca, all of the "/Alpaca Client Device Base Classes/TYPE Base Class.cs"
source blocks any attempt to launch the (from):
https://ascom-standards.org/api/?urls.primaryName=ASCOM%20Alpaca%20Management%20API#/HTML%20Browser%20User%20Interface
..........
Alpaca Management URLs are all of the form http(s)://host:port/path where path is one of the method names below. e.g. for an Alpaca interface running on port 7843 of a device with IP address 192.168.1.89:

The main browser setup URL would be http://192.168.1.89:7843/setup
..........
If SetupDialog is invoked on a driver, in Alpaca if the device is not 'Connected' then the attached is what I see.
It looks very specific to the device
[image]

I am still at a loss why the framework chose to do a duty that was relegated to the deive drivers themselves in legacy COM.

This is -not- 100% compatible.

How is it recommended a client program get to an Alpaca device that -does- support /Setup when connected, is the user required to memorize the URL/port they see in the above screen shot and type it into a browser manually?

Gene

astrogene1000, 2023-12-13:
Hi Peter,

Sounds like a plan!
" "Alpaca Web Configuration" buttons be available even when the dynamic client is connected and it is not permissible to change any of the configuration parameters shown in the image above?"

**Yes, please,** this would allow the Alpaca endpoint to make the decision, like in COM for 'SetupDialog' invocation.

May want to allow the 'Trace' options also :-) Possible to bring those out?

Gene

Peter-Simpson, 2023-12-14:
Hi Gene,

OK, I'll make those changes but it will likely be a while before they see the light of day because we are in the middle of complex revisions for a new Platform 7 release that incorporates the async interface changes we consulted on recently.

Best wishes, Peter

astrogene1000, 2023-12-14:
Thanks you Peter!

Want to keep this open until changes hit production builds?

One question please, is there any concise directions for building out ASCOM from GIT fork?

Gene

Peter-Simpson, 2023-12-18:
Hi Gene,

Probably best to leave this open so I don't forget! 😁

If you mean build as in creating an installer, there is a description of how to build a build machine in the "/Build Process" folder in the repo. This is a complex business and requires a code signing certificate as well as the purchasable InstallAware product.

OTH, if you just mean to compile the Platform code, the Release/AnyCPU configuration on the master branch should work if you update to include the commits I made today.

Best wishes, Peter

Peter-Simpson, 2024-08-03:
This is now available for testing in the Platform 7 release candidate: [https://github.com/ASCOMInitiative/ASCOMPlatform/releases](https://github.com/ASCOMInitiative/ASCOMPlatform/releases).

Best wishes, Peter

astrogene1000, 2024-08-19:
Hi Peter,
Saw the announcement for RC5, did not have chance to load it but did go looking at source for tag V7 RC5 focuser.cs and do not see any changes in SetupDialog. All files show last changed date 8 months ago.
I do see you original update for Alpaca Camera Devices but so far this is the only one updated?

https://github.com/ASCOMInitiative/ASCOMPlatform/tree/v7.0.0-rc.5/ASCOM.Utilities/AlpacaDynamicClients/Dynamic%20Client%20Local%20Server/Devices/

Thanks!

Gene

Peter-Simpson, 2024-08-19:
Hi Gene,

Thanks for the reminder! I've now added support for the other device types to the develop branch and built a new version: [RC 6 test build](https://download.ascom-standards.org/beta/platform7/AscomPlatform700.rc6.4467.exe).

Does this now do what you needed?

Best wishes, Peter

astrogene1000, 2024-08-20:
Hi Peter,

I am flying today, should get chance over next day or two, thank you!

Has RC6 been tagged and checked in or private currently?

One part of the original issue I believe was a direct call to SetupDialog by a higher level application, I give the RC6 a test fly.

Thank you again!

Gene

Peter-Simpson, 2024-08-20:
Hi Gene,

 

Thanks for testing this!. The first part of the fix was for the SetupDialog call from a higher application as you say, I just extended it so that the same happened when called from the Chooser.

 

The changes are all pushed to GitHub on the “develop” branch at the moment, I’ll copy them to “main” and add a tag when I make a more formal release of RC6.

 

Best wishes, Peter

astrogene1000, 2024-08-20:
Hi Peter,

I will have to see how it reacts, not quite following the SetupDIalog code base in dev branch as seems to direct to SetupConnectedForm if connected and if not connected to  Server.SetupDialogue<AlpacaFocuser>

`
        public void SetupDialog()
        {
            if (connectedState) // Only display Alpaca configuration buttons if already connected
            {
                using (SetupConnectedForm connectedForm = new SetupConnectedForm(TL))
                {
                    connectedForm.DeviceNumber = state.RemoteDeviceNumber;
                    connectedForm.DeviceType = state.DeviceType.ToDeviceString();
                    connectedForm.HostIpAddress = $"{state.ServiceType}://{state.IpAddressString}:{state.PortNumber}";
                    connectedForm.ShowDialog();
                }
            }
            else // Show dialogue
            {
                AlpacaFocuser newclient = Server.SetupDialogue<AlpacaFocuser>(state, TL);

Thanks!

Gene`

Peter-Simpson, 2024-08-20:
Hi Gene,

 

Instead of the normal Dynamic Client setup dialogue you will get this “cut down” version that provides access to the Alpaca device’s HTML configuration pages and enables tracing to be enabled / disabled. These are the only two things that are safe to do when the Alpaca device is connected.

 

Best wishes, Peter

astrogene1000, 2024-08-20:
Hi Peter,

Ok, but the compatibility with legacy is not fulfilled.

In the discussion above you had said
"Many drivers that read their configuration at start-up prevent configuration changes when connected in order to protect the integrity of the driver,"

Yes, it may be true 'many' drivers but my original premise is not 'all' drivers.

The target device -could- make the decision to not run the SetupDialog no matter how many clients are connected, or is this totally incorrect?
In COM it is true but I do notice all current example templates bail out.

Thank you for your time in this, the issue can probably be closed as not sure we will see eye to eye and I do not want to distract to the EXCELLENT work the ASCOM team does :-)

I think we need a lot more Alpaca users :-)

Gene

Peter-Simpson, 2024-08-20:
Hi Gene,

 

I’ve copied the confirmatory text below that you posted earlier in this thread. I believe that I have delivered what you asked for, so it’s disappointing that you now say it isn’t what you were expecting.

 

The changes I made make “Alpaca Web Configuration” buttons available when the SetupDialog is called and the device is connected.

 

If the Alpaca device doesn’t allow the configuration to be changed while connected, it can simply say this on the returned HTML configuration page and not display buttons to change the configuration.

Regards, Peter

 

Hi Peter,

Sounds like a plan!
" "Alpaca Web Configuration" buttons be available even when the dynamic client is connected and it is not permissible to change any of the configuration parameters shown in the image above?"

Yes, please, this would allow the Alpaca endpoint to make the decision, like in COM for 'SetupDialog' invocation.

May want to allow the 'Trace' options also :-) Possible to bring those out?

Gene

astrogene1000, 2024-08-21:
Hi Peter,

Sorry for any confusion, I had missed the access to setup via 
"The changes I made make “Alpaca Web Configuration” buttons available when the SetupDialog is called and the device is connected."

let me run RC6 and I had missed this capability!!!

Thank you again!

Gene

astrogene1000, 2024-08-22:
Hi Peter,

Basic non connected setup works as expected.

Are there any new API's beyond:
https://ascom-standards.org/api/#/ASCOM%20Methods%20Common%20To%20All%20Devices
https://ascom-standards.org/api/#/Focuser%20Specific%20Methods

ASCOM Diagnostics, when try to connect, gives me this (I also see same in NINA)
Create              Creating device
Connected           Connecting to device
Connected           Trying Connected
Connected           Trying Link: 'T' is an invalid start of a value. Path: $.ClientTransactionID | LineNumber: 0 | BytePositionInLine: 23.
Error               System.Exception: 'T' is an invalid start of a value. Path: $.ClientTransactionID | LineNumber: 0 | BytePositionInLine: 23.
   at System.Dynamic.ComRuntimeHelpers.CheckThrowException(Int32 hresult, ExcepInfo& excepInfo, UInt32 argErr, String message)
   at CallSite.Target(Closure , CallSite , ComObject , Boolean )
   at System.Dynamic.UpdateDelegates.UpdateAndExecute2[T0,T1,TRet](CallSite site, T0 arg0, T1 arg1)
   at System.Dynamic.UpdateDelegates.UpdateAndExecute2[T0,T1,TRet](CallSite site, T0 arg0, T1 arg1)
   at ASCOM.DriverConnect.ConnectForm.btnConnect_Click(Object sender, EventArgs e) in C:\ASCOM Build\Export\Diagnostics\ASCOM.DriverConnect\ConnectForm.cs:line 251
Dispose             Disposing of device
Dispose             Completed disposal
ReleaseComObject    Releasing COM instance
ReleaseComObject    Completed release. Count: 0
GC Collect          Starting garbage collection
GC Collect          Completed garbage collection

I am going to add some debug to my device to see inside but 'LINK' API?

Gene

Peter-Simpson, 2024-08-22:
Hi Gene,

 

The interface changes are listed in the Developer’s Help file that you will find in your Start / ASCOM Platform for Developers folder.

 

Link is the original property used to connect Focuser interface version 1 devices. At this point Diagnostics doesn’t assume a particular interface version and, if the normal Connected method fails, as happened here, it will try the Link property as a fallback strategy. Focuser is the only device that ever used a property other than Connected to connect to the hardware.

 

Regards, Peter

astrogene1000, 2024-08-22:
Hi Peter,

I believe I found where the "T" is coming from.
The order of the parameters changed in the PUT for Connected (maybe all API's have changed order?)

The order I am seeing is "Connected, ClientID, ClientTransactionID"
I do not do parsing of the parameter name but blindly assumed the order shown in the API doc
"ClientID, ClientTransactionID, Connected"

As seen incoming:
**URI: /api/v1/focuser/0/connected
Method: POST
Arguments: 3
Connected: True
ClientID: 48412
ClientTransactionID: 2**

So the "T" is coming from my grab of "Connected: True" assuming it is the "ClientTransactionID" and returning the "True" as its value.

Be advised, the device on the end of the wire may be a small less powerful micro and string parsing is expensive.

Gene

Peter-Simpson, 2024-08-22:
Hi Gene,

 

HTTP/REST/JSON do not require clients to send parameters in any particular order, it is sufficient for clients to send them in any order. Your device shouldn’t have a dependence on parameter ordering for the reason that you’ve already discovered.

 

If you need further help with your development, it would be better to post questions on the Developer Forum <https://ascomtalk.groups.io/g/Developer>  rather than extend this GitHub issue into a generalised help topic.

 

Have you tested the setup dialogue with the device connected? I’d like to close this issue or fix any remaining bugs.

 

Regards, Peter

astrogene1000, 2024-08-22:
Hi Peter,

The original "DialogSetup" ask looks good in RC6, Super THANKS!

I do understand "HTTP/REST/JSON do not require clients to send parameters in any particular order".

I will say it was an arbitrary developer decision though to reorder the sent parameters  changing from 6 to 7.

If I may make a suggestion, update the API doc's to indicate the differences between 6 and 7 for folks with limited processing horsepower on the server side.

I would surmise all 'PUT's that show a 'required' parameter in 7 now have the 'required' as the first parameter, besides Connected I also see it in Focuser Move.

Thank you very much for doing the path to DialogSetup for devices that can support it!

Gene

Peter-Simpson, 2024-08-22:
Hi Gene,

*	I will say it was an arbitrary developer decision though to reorder the sent parameters changing from 6 to 7.

The JSON serialiser we use was changed between Platform 6 and 7 but this was after much thought and was definitely not an arbitrary decision. Beyond that high level decision, we have no control or interest in how the serialiser does it job. 

 

I am confident that the serialiser we use is generating HTTP JSON in accordance with Internet protocol standards, which do not require a particular or consistent parameter order. If any other Alpaca client connects to your device, it may use a completely different parameter order over neither you nor we have control.

 

Regardless of the amount of processing power available, your device will need to handle the flexibility allowed by Internet protocols.

 

I’m glad the change I made for you is working OK, I think this issue can now be closed.

 

Regards, Peter

astrogene1000, 2024-08-22:
Thank you Peter, I will close this out!

closed 2024-08-22 by astrogene1000