ASCOMInitiative/ASCOMPlatform #71: Platform 7 System.IO.FileNotFoundException for Apogee FW

ramonefoster, 2024-07-15:
Got System.IO.FileNotFoundException. Could not load file or 
assembly 'X' or one of its dependencies.
Exception when trying to connect Apogee FW to new Platform 7. Driver works fine in Platform6.6 but something might be broken for the 7.0 version of the platform.

Peter-Simpson, 2024-07-16:
Hi Ramone,

I found that some .NET Framework assembly re-direction policies were being installed in the wrong place and this may explain why the exceptions assembly seems to be missing when it is in fact installed correctly in the global assembly cache. 

The redirection policies tell .NET that requests to load earlier versions of the assembly should use a later version instead. If they are missing, requests to load the assembly will fail and the assembly will be reported as not present .

Please can you try this build to see if it fixes the issue? [Download link](https://download.ascom-standards.org/beta/platform7/AscomPlatform700.rc1.4432.exe)

Best wishes, Peter

ramonefoster, 2024-07-16:
Thank you for your fast response, Peter. I just checked and everything looks fine now.

Cordially, Ramon

closed 2024-07-16 by ramonefoster

Peter-Simpson, 2024-07-16:
Great, thanks very much for testing and for letting me know!

Best wishes, Peter

commit eee254af (2024-07-16): Revert redirection policies to CLR2, which had unintentionally been converted to CLR4 during Platform 7 development, fixes #71.

The policies need to be in the CLR2 GAC in order to be effective so have to be compiled as CLR2.