ASCOMInitiative/ASCOMPlatform #72: ASCOM 7 RC4, Install problem, sees old developer tools and can not uninstall it

han-k59, 2024-08-02:
Can not install Ascom7, RC4. See screenshot.  Developer tools are NOT installed. Anything in registry I could remove?

Han

[image]

han-k59, 2024-08-02:
I can install the following ASCOM versions
6.4,
6.61,
6.6.2beta.3677
6.6.2.4294

but not 
7RC1
7RC4

han-k59, 2024-08-02:
The commandline option does not help either:
   /removeascom command line parameter

Peter-Simpson, 2024-08-02:
Hi Han,

The issue is that the Platform 6 Developer Tools application is recorded as being installed on your PC but its uninstall program is missing or not working for some reason. From my experience, the application can no longer be uninstalled automatically when this happens.

The Platform 7 installer provides all the files that were present in the Platform 6 Developer Tools so I am building a revised installer that will quietly remove the developer tools install record when the application can't be uninstalled. This should allow the install to proceed normally for you.

I'll post a link to the new installer in a couple of hours when its built and tested.

Best wishes, Peter

Peter-Simpson, 2024-08-02:
Hi Han,

Here is a revised installer that should fix the issue you saw: [AscomPlatform700.rc4.4449](https://download.ascom-standards.org/beta/platform7/AscomPlatform700.rc4.4449.exe)

Best wishes, Peter

Peter-Simpson, 2024-08-03:
Please can you confirm whether this has fixed the issue.

Best wishes, Peter

han-k59, 2024-08-03:
Yes this version installs fine. Thanks.

Sorry I could not respond earlier. 

Cheers, Han

Peter-Simpson, 2024-08-06:
Thanks Han, closing this as fixed.

best wishes, Peter

closed 2024-08-06 by Peter-Simpson