ASCOMInitiative/ASCOMPlatform #73: Problem using an Alpaca device via COM interface

han-k59, 2024-08-24:
Got the following problem running win7.1:

[image]

I'm trying to access an Alpaca device via COM. I'm running Win 7.1

Alpaca device is the "Sky simulator ASCOM"

Han

han-k59, 2024-08-24:
The problem doesn't occur in Win10.

Peter-Simpson, 2024-08-25:
Hi Han,

Thanks for reporting the issue, which I have been able to replicate. It arises because the Dynamic Client manager was migrated to .NET Core and requires support libraries that are not present in Windows 7. I saw a prompt to download and install the runtime library when I replicated the problem, but the resulting Dynamic Client was not registered, exactly as you found.

You can delete the bad Sky Camera client using the Manage Devices (Admin) option in the Chooser Alpaca menu as shown below.

I've just reverted the Dynamic Client manager to .NET Framework so you should no longer see this issue. Please can you try this version and let me know if it fixes the issue. [Fixed Platform 7 Installer](https://download.ascom-standards.org/beta/platform7/AscomPlatform700.rc6.4483.exe).

Best wishes, Peter

[image]

han-k59, 2024-08-25:
Hi Peter,

Thanks. That fixes the problem.

I noted a typo error in the the last message of the installation. I assume it should be successful. See screenshot.

[image]

Cheers, Han

han-k59, 2024-08-25:
The installation complete message in Win10 used the word Successfully.  Weird. 

[image]

daleghent, 2024-08-25:
Must we really hold back progress and cater to versions of Windows that Microsoft themselves has abandoned?

Peter-Simpson, 2024-08-25:
Hi Dale,

The Platform is rooted in .NET Framework and I don't see that changing if we maintain backward compatibility, due to the need to support .NET CLR2 applications. However, for a simpler end-user experience with our .NET Core apps, we've taken the approach of not requiring runtime installations by publishing apps like ConformU and the OmniSims self-contained. 

I forgot about this when migrating the Dynamic Client manager app, hence the error that Han experienced. I suspect that .NET Framework CLR4 will have plenty of maintenance life, so chose to revert rather than handle the additional complexity of publishing and the resultant increase in installer size, which is already 3 times larger for Platform 7 than Platform 6 due to the self-contained decision.

The Platform's Framework legacy is why we created the ASCOM Library using current modern technologies. Of course,  my choice can be debated, but maintaining backward compatibility and ensuring maximum ease of use are two of our core priorities.

Best wishes, Peter

Peter-Simpson, 2024-08-25:
Hi Han.

The "Installation unsuccessful" image you posted is worrying because it tells me that something went wrong that was not recorded as an error by the installer.

Please can you run ASCOM Diagnostics program as described [here](https://ascomtalk.groups.io/g/Help/wiki/18368) then zip up all the files and sub-folders in your Documents\ASCOM folder and post the zip archive here so I can try to work out what happened.

Many thanks, Peter

daleghent, 2024-08-25:
> Hi Dale,
>
> The Platform is rooted in .NET Framework and I don't see that changing if we maintain backward compatibility, due to the need to support .NET CLR2 applications. However, for a simpler [...]

There's a distinct difference between using an older framework because it happens to still be maintained, and continuing to use it in order to placate the needs of those who choose to stay on versions of Windows that Microsoft has forsaken. You are now making the decision to couple the reason to maintaining usability with EOL'd OS versions rather than natural framework evolution, which implies you will continue to do the same for other components in the ASCOM stack - thereby holding the entire platform back as technology moves forward. If some other ASCOM component fails to work on Windows 7, will you also revert it or re-implement it to work under the older Framework, too?

It's worth noting that Windows 7(.1) also has trade-offs under .NET Framework. For example, Windows 7.1 systems cannot support TLS 1.3 through .NET's HTTP client. Since most sites have moved to supporting TLS 1.3 and a large subset of those have deactivated TLS 1.1 and 1.2 support, we've found that this leaves Windows 7.1 users of NINA without a way to use such sites, which includes NASA websites (the US Gov. mandated TLS 1.3 only support some years ago.) This just underscores the need to move on from Windows 7 as a general policy and bending over backward, even if doing so is "easy", removes reasons for users to get with the program and move on themselves. They cannot stay on Windows 7 forever, and ASCOM should not be placating that.

han-k59, 2024-08-25:
Hi Peter,

Attached the log of the diagnostics

[Logs 2024-08-25.zip](https://github.com/user-attachments/files/16740777/Logs.2024-08-25.zip)

Han

Peter-Simpson, 2024-08-25:
Hi Han,

`Please can you run ASCOM Diagnostics program as described [here](https://ascomtalk.groups.io/g/Help/wiki/18368) then zip up all the files and sub-folders in your Documents\ASCOM folder and post the zip archive here so I can try to work out what happened.`

Your zip archive doesn't include the contents of the Documents\ASCOM folder where the Platform Installer logs are located.  Please can you zip up and post the contents of Documents\ASCOM. The installer logs I need are named similarly to this: ASCOMPlatform7Install487907450880.txt

Regards, Peter

han-k59, 2024-08-25:
Here they are:
[ASCOM install logs.zip](https://github.com/user-attachments/files/16741412/ASCOM.install.logs.zip)

Peter-Simpson, 2024-08-26:
Hi Han,

Thanks for the logs, unfortunately they do not show why you got the partially completed dialogue without even the rectangular black box that should contain the message. 

I have inspected the installer code and can see no path that leads to an unsuccessful installation message but does not include text explaining why. I have added further logging code to aid diagnosis should this happen again, but I'm unable to replicate this issue here, the installer works fine on my test Win 7 64bit system.

The installer log suggests that the OmniSims may not have registered correctly, but if you don't use the Platform simulators this shouldn't cause you an operational issue.

The Diagnostics log only shows one error, which was caused by an unreadable registry key in the Green Swamp Server COM registration. Diagnostics identified this but then tried to close the non-existent registry key, which caused the exception. This is a Diagnostics bug that's fixed in the installer linked below.

I would like to draw your attention to the Platform 7 Operating System Compatibility statement that you will find in the Help file, copied below. This has been in place since the start of Platform 7 development and shows that support for Windows 7 is on a best endeavours basis. At present I think that I've gone as far as I can to help you with this issue.

This installer should fix the Diagnostics error and will hopefully provide a working Platform 7. 
[Platform 7 with Diagnostic fix and more installer logging](https://download.ascom-standards.org/beta/platform7/AscomPlatform700.rc6.4486.exe).

Best wishes, Peter

[image]

han-k59, 2024-08-26:
Thanks Peter.

My intension was to help the ASCOM initiative with testing, not to fix my problems.

I run the latest installer but still an unsuccesfull message. I have attached the install logs and diagnostics log.

Cheers, Han

[ASCOM log and diagnostics.zip](https://github.com/user-attachments/files/16747608/ASCOM.log.and.diagnostics.zip)
[image]

4greg2, 2024-08-26:
I’ve been thinking about saying something but I gave in and thought I’d
toss in my 2cents worth
(Probably an Oz expression)

I agree with Dale. We can not advance just because people don’t like or are
scared of change (or have been burned by updates)

BUT, I feel for people with older drivers and wonder why we don’t have a
legacy transcoder (don’t shot me with the name - particularly as I know
this is a large US audience - BTW I’m Oz)

Like we have an Alpaca interface (hmm, I hope I used the correct term), why
no advance and have some form of connector for those drivers on older gear,
perhaps the vendor has disappeared or whatever.

Let the legacy survive but advance the others. I want the advances, the new
ideas, the updates and the whatever’s

While I’m approaching mt twilight years, I’m not a software dinosaur or
scared of change. New bugs keep my mind active:-)

PS: Dale has contributed bucket loads of effort to the hobby. He has earned
the right to add to the discussion and I support him voicing his approach
to the future ASCOM direction.

Peter-Simpson, 2024-08-26:
Hi Han,

Thanks for reporting the issue, I do appreciate it. The logs show that the OmniSim installation failed again and that the installer was trying to display a final message showing what went wrong. However, that message in a black rectangular box is not shown by your PC for some reason. I've attached an image showing the black box and a successful outcome message.

Your finish dialogue appears different to mine with square corners instead of rounded corners so there must be some sort of visual style difference on your PC.

I can't find any diagnostic information on error 0x80008082 (-2147450750) to explain why the OmniSim installation fails on your PC, despite the installation working OK on my test PC.

I'm pleased that Diagnostics now passes all tests.

Best wishes, Peter

[image]

Peter-Simpson, 2024-08-27:
Hi Greg,

`I’ve been thinking about saying something but I gave in and thought I’d toss in my 2cents worth...`

I absolutely agree that we need to keep up to date with technology wherever possible. For example, Platform 7 incorporates ASCOM Remote and the Omni Simulators, which are both .NET 8 applications. All the Platform .NET Framework CLR4 components use .NET Framework 4.7.2 or later. We have a number of components that have to be .NET Framework CLR2, which can't be updated to CLR4 or .NET Core without breaking existing applications. They have to stay as is.

To mitigate this risk we created the ASCOM Library, which current developers can use in .NET Framework 4.5 onward and .NET Core 1.0 onward applications / drivers / Alpaca devices. This enables them to use ASCOM components in the most modern development technologies of their choice.

My .NET Core implementation of the Dynamic Client manager was dependent on a .NET Core runtime being installed on a users machine.  If the runtime  wasn't installed, users would get Han's experience of a Microsoft request to install the run-time through a browser GUI and would still find that they got a failed Dynamic Client installation that had to be manually deleted. 

Avoiding this experience of having to pre-install run-times is why we package our .NET Core apps as stand-alone builds.
 
The options I saw were :

1. Publish the Dynamic Client manager as a stand-alone app and include all the support files in the installer thus increasing its size further and changing the build process to integrate the published version.

2. Revert back to .NET Framework to save increasing the installer size further while keeping the new SDK project format to use modern tooling. This avoided having to revise the complex Platform build process to publish and incorporate the larger application at this late stage in the release process.

I chose 2 as the lower risk option because it makes no difference to the end user experience and still leaves the door open to update at a later date. That was my value judgement in this situation with these particular circumstances. Dale would not be correct to infer that this means all update decisions will be made similarly.  each decision is judged on its merits.

`Dale has contributed bucket loads of effort to the hobby. He has earned the right to add to the discussion and I support him voicing his approach to the future ASCOM direction.`

I agree and I feel the same applies to me, having spent 15 years maintaining the ASCOM Platform for the benefit of the astronomy community.

Best wishes, Peter

daleghent, 2024-08-27:
I think the point was missed, Peter. This is about which versions of Windows is supported, not whether Framework or Core is used.

The root cause of Han's issue was that he was running on Windows 7.1, a version of Windows that has been de-supported by Microsoft for (as of this writing) 4.5 **years**. You reverted a change that broke only on Windows 7, and the evidence says it worked fine on newer versions of Windows. Therefore the motivation for your reversion _was to support Windows 7_. 

When it comes to decisions around which version of Windows is supported, that's not something one can take on a case-by-case basis. The whole system must Windows back to version X, or it doesn't. Thus, by making this reversion to suite Windows 7, despite it not affecting newer versions of Windows, you're pretty much making the proclamation of "We will support Windows 7." What I'm urging you to do is to say which version of Windows you will support with Platform 7. If that includes Windows 7, I would ask you to reconsider and instead follow Microsoft's own lifecycle schedule. Supporting rotting OS versions that people choose to stay on for highly idiosyncratic reasons is not a good policy.

han-k59, 2024-08-27:
I do not want to start an endless discussion. But sure ASCOM Alpaca should run on Win11 and Win10.  If it can run on Win7 that is and nice to have bonus. I'm not the only one still having an old Win7 system in my astro shed. If ASCOM 7 does not run I can just install ASCOM 5 or 6 instead. Note that Win7 is still running on more then 3% of the PC's wordwide. So it is not rotten but just old. And I do not want to buy an overkill multimedia gaming PC for my mount control to do some something as simple  ASCOM communication which commands go finally through an RS232 serial link at 9600 baud to my mount. Yes my telescope has no USB3.1, fiber optic or ethernet connection.  Is also has no cloud connection.

As a comparison, my own written astro programs run natively on Windows, Linux, macOS, Android with i386, x86-64, M1, arm64 processors because users have all these systems. I try to support all the hardware of my users and are less dictated by a lifecycle schedule and fully independent of .NET versions :).

Just my toughts.

daleghent, 2024-08-27:
By your measure, should ASCOM bend over backwards to support Vista? How about XP? Where or when does it end? Declaring, even tacitly, the minimum supported versions of the underlying OS dictates the technological possibilities for the platform. Its capabilities and direction will be tied to the lowest supported version., whatever it is This would extend to items such as the driver templates and those producing drivers and the people producing drivers. I think that folks would largely follow that cue and say "well if ASCOM supports Windows 7, I suppose my driver should as well." How are these folks going to test their software to make sure it works on an OS that isn't available anymore?

And, please, don't make obviously outlandish claims. Modern versions of Windows and imaging apps run fine for DSO imaging on a wide range of low-power PCs. You don't need anything like a "gaming PC" to do imaging with a modern OS and software stack. In fact, it's very likely that you would be easily able to find a low-power (<= 25W TDP) system that has far more compute power than whatever system you're using to run Win7.

nightskywatcher, 2024-08-27:
Gee I hate to chime in because I’m so grateful to Peter and Dale…

But…

There is no excuse for running anything earlier than Win 10 IMHO because of the cost of a new computer:

https://www.amazon.com/MeLE-Mini-Quieter-4C-Astrophotography/dp/B0CP3YL6J7

$200, runs like a scared rabbit.

I’m a programmer and an astro guy and while I’m all about backwards compatibility, COM/DCOM was the bee’s knees of the 90s and is way long in the tooth.

I totally agree with Peter that Platform 7 should be the complete focus of the moment, but perhaps a subgroup of developers can start talking about what architectures are available now and involve the vendors, especially the Alpaca contingent like OPTEC. Maybe 2-3 years down the road.

Win 7 machines make great remote desktop clients,  so you invest $200, and freeze a version for those who don’t want to move forward… but end support or let a 3rd party pick up maintenance (like WinNT 3.51 kernel purchased as CITRIX) of Legacy ASCOM.

I know it’s easy for someone who does not contribute to comment, but thanks for listening anyway.

Regards

David Diaz
Grateful ASCOM user

amckeown123, 2024-08-28:
I would second the sentiment being expressed here.  If you can afford
astronomy you can afford a new computer and upgrading to windows 11 is
possible on potatoes computers.  I do this for my friends all the time.
And the older software run just fine in compatibility mode.   Also you
should look at getting up to vs 2022 as that is very stable and will go
back to windows 8
You will need to update the .net to a later version though.

michealroberts, 2024-08-28:
For RS version 7, I would recommend publishing a list of OS versions that are officially supported.

I would always go by this one simple rule: “if the OS version end of life date proceeds the date of publish for the new software, there should be no guarantee of compatibility or support”.

Announced:

Windows 7 - January 10, 2023.
Windows 10 - October 14, 2025.
Windows 11 - No date announced yet.

I think that should give the ASCOM team clear confidence on which OS to officially support based off of the above rule. :)

daleghent, 2024-08-28:
35 folks watch this repository, and the default is for "all activity". Thanks for considering the question since a major rev to the Platform is a rare opportunity to set or clarify important policy for years to come.

DanielVanNoord, 2024-08-30:
I'm going to add that we have two separate issues here, which we may want to split into separate threads if there is further discussion:
1. That the dynamic client manager required .Net Core to be installed on the machine in order from them to work. The installer does not require this. Reverting them to .Net Framework seems to me to be a completely acceptable solution as they are Windows only. I believe that one of the current goals of the platform is to minimize the amount of external dependencies that we need to install. This was discovered on a Windows 7 machine but is an issue for all versions of Windows, including 10 and 11. Similarly a lot of COM stuff is likely to need to stay on .Net Framework because of a bug in .Net Core (and 5+).
2. The second issue is what the support policy for EOL operating systems should be. This is separate, fixing the original issue isn't a Windows 7 vs 10 issue. From a quick test this was breaking just as badly in one of my Windows 11 VMs.

Peter-Simpson, 2024-09-07:
Hi all,

In response to the issues evidenced in this thread and your feedback on technology obsolescence, we have decided that Platform 7 will only be supported on Windows 10 (Build 1809) October 2019 and later operating systems.

The Platform installer in release candidate 7 now enforces this decision and does not allow installation on Windows 7.

It is always regrettable when we have to make decisions like this, which "leave people behind". But this has to be balanced by the Windows 7 OS being 15 years old, and that it has not been supported by Microsoft four four years.

Thankyou for the helpful feedback that has resulted in this change.

Best wishes, Peter

closed 2024-09-07 by Peter-Simpson

daleghent, 2024-09-08:
Thank you Peter. This is a good baseline seeing as how Windows 8 is also EOL'd.