ASCOMInitiative/ASCOMPlatform #74: RC5 and GS server 1.0.8.5

han-k59, 2024-08-24:
I have just updated to GS server 1.0.8.5. At the moment it doesn't work with RC5. The old GS server version 1.0. 5 did work with RC5, so likely a problem of GS server.

han-k59, 2024-08-24:
I get errors like this one:
15:00:35 ASCOM.GS.Sky.Telescope: Connection error: The calling thread must be STA, because many UI components require this.

han-k59, 2024-08-24:
I have raised an issue at GSS:

https://github.com/rmorgan001/GSServer/issues/73

Describe the bug
I have a Win 7.1 system and ASCOM 7, RC5 installed.

GSS version 1.0.5 worked fine.

Upgraded to 1.0.8.5 and also tried 1.0. 8.1 but both throw an error if GSS is started from an application (CCDCiel and Sky simulator for ASCOM). The error is as follows:.

ASCOM.GS.Sky.Telescope: Connection error: The calling thread must be STA, because many UI components require this.

If I first start GSS manually using the desktop icon prior to starting the other applications then it all runs and connect normal. So starting GSS from the other applications is a problem.

Han

Peter-Simpson, 2024-08-25:
Hi Han,

I'm closing this ASCOM Platform issue because the GSS team have identified the problem as being with their code and not with the Platform.

Best wishes, Peter

han-k59, 2024-08-25:
Hi Peter,

thnaks. That fixes the problem.

I noted a typo error in the the last message of the installation. I assume it should be successful. See screenshot.

[image]

Cheers, Han