ASCOMInitiative/ASCOMPlatform #75: Publish to winget?

RReverser, 2024-08-27:
[Windows Package Manager](https://learn.microsoft.com/en-us/windows/package-manager/) aka [`winget`](https://github.com/microsoft/winget-cli) became the de-facto installation tool and store for 3rd-party apps on Windows in the last years.

It's particularly convenient for automatically managing app updates, migrating from one computer to another and so on.

Would it be worthwhile / possible to publish ASCOM tools (ASCOM Platform itself, simulators and perhaps ConformU) to the winget  store?

Peter-Simpson, 2024-08-27:
Sorry Ingvar, I'd prefer that we don't start something completely new like that right now. Our number one priority is to get Platform 7 fully functional, stable and out the door.  I don't want any distractions from that at the moment.

It's a great idea and certainly something to explore when the Platform 7 dust settles, but I'd like to park this for now.

Best wishes, Peter

RReverser, 2024-08-27:
Absolutely; just wanted to throw it out there as I recently migrated to a new laptop and could easily install all the same apps by exporting a winget list and importing on the new machine, but a few apps like ASCOM had to be downloaded manually.

Certainly no rush on it though.