ASCOMInitiative/ASCOMPlatform #78: ASCOM 7  Telescope Simulator Setting problem

tsatatwer, 2024-10-17:
I have installed the ASCOM Platform 7.0 I did access the web UI and configure the simulators and then checked unlock after that if I choose a device and when I click setting I get an error see screenshot in attachment .Thanks
[image]

Peter-Simpson, 2024-10-17:
Hi,

The OmniSimulators are now the default Platform simulators and are configured through their browser interface. When you see the message above, you need to start a browser on http://localhost:32323 where you can change the simulator configuration as required.

The message above is clearly not sufficiently informative, so iIwill add further text to make it clear how to configure the OmniSimulators.

Best wishes, Peter

tsatatwer, 2024-10-17:
Hi Peter,

I know the **OmniSimulators**  are configured through **http://localhost:32323/** and I did already , the question is then for these **OmniSimulators** configuration is done only through web interface ???? there is no other way to configure them (Setting Window) ???

And are the **Telescope Simulator for .NET** and **Camera V2 Simulator for .NET**  **OmniSimulators** ???
Thanks

Peter-Simpson, 2024-10-17:
Hi,

Yes, the OmniSimulators are only configured through the web interface. 

I'm not quite sure what you are asking in your last sentence, but if you are asking whether it is possible to restore the original simulators as the default Platform simulators, the answer is "yes", using the [Diagnostics tool](https://www.ascom-standards.org/help/developer/html/0a23ff52-a935-4cd7-a693-9e3b90f7df29.htm).

Best wishes, Peter

tsatatwer, 2024-10-17:
Hi Peter,
You have answered my question clearly  ,  so from what I understand that device configuration through the web interface is only for OmniSimulators that is .
And as you suggested please make the message more  sufficiently informative .

Thanks

Peter-Simpson, 2024-10-17:
The new message is:

```
The Device Simulator is configured through the Alpaca Web UI by starting a browser and navigating to: http://localhost:32323

Press the OK button when you have finished making changes through the Web UI.
```

Best wishes, Peter

closed 2024-10-17 by Peter-Simpson