ASCOMInitiative/ASCOMPlatform #79: Connecting Alpaca camera via ASCOM Remote Server bridge fails the first time,

han-k59, 2024-11-25:
I'm testing my "Sky simulator for Ascom and Alpaca" using Ascom 7.0. Via the ASCOM Remote Server there is bridge from COM to Alpaca. So with Ascom 7 installed you can access an Alpaca camera  as a COM device in programs CCDCiel or Maxim DL

This works fine for AscomPlatform662.4294.exe. But for  AscomPlatform701.4630.exe the connection to the camera fails the first time but it works the second time . 

I get an error in CCdCiel like:
18:33:36 ASCOM.AlpacaDynamic8.Camera: Connection error: Value not set Connected get - no value has been set.

In MaximDL the same happens. First time it fails. The second time it connects.

This is what I see in the Alpaca log of the simulator:
```
■■■■■■■■■ Duration: 90.5 sec.
21:48:33.030  PUT/api/v1/camera/0/connected      =>   {"ClientTransactionID":0,"ServerTransactionID":371,"ErrorNumber":1026,"ErrorMessage":"Value not set Connected"}

■■■■■■■■■ Duration: 6.6 sec.
21:48:39.658  PUT/api/v1/camera/0/connected   Connected=False&ClientID=24838&ClientTransactionID=2   =>   {"ClientTransactionID":2,"ServerTransactionID":372,"ErrorNumber":0,"ErrorMessage":""}

21:48:40.985  PUT/api/v1/camera/0/connected   Connected=True&ClientID=5549&ClientTransactionID=3   =>   {"ClientTransactionID":3,"ServerTransactionID":373,"ErrorNumber":0,"ErrorMessage":""}

21:48:41.016  GET /api/v1/camera/0/connected?ClientID=5549&ClientTransactionID=4 => {"Value":true,"ClientTransactionID":4,"ServerTransactionID":374,"ErrorNumber":0,"ErrorMessage":""}

21:48:41.047  GET /api/v1/camera/0/driverinfo?ClientID=5549&ClientTransactionID=5 => {"Value":"Camera Sky Simulator.","ClientTransactionID":5,"ServerTransactionID":375,"ErrorNumber":0,"ErrorMessage":""}

21:48:41.063  GET /api/v1/camera/0/driverversion?ClientID=5549&ClientTransactionID=6 => {"Value":"v1","ClientTransactionID":6,"ServerTransactionID":37
```

Note the missing **"Connected=True"** parameter in the first log line.  Then I disconnect and reconnect and it works

Any idea what goes wrong in Ascom 7 ?  Ascom 6.6 it connect immediately by the first request.

cs Han

The error message of MaximDL. 

[image]

Peter-Simpson, 2024-11-26:
Hi Han,

I can reproduce this here but it may be an issue with your implementation. The error message is coming back from the sky simulator in response to the set connected true Alpaca command from the dynamic client, see text below.

I tried sending a PUT connected = true from my REST client, completely bypassing all other ASCOM code, and it received the same error, see image below.

Is there somewhere in your connect code where you can return an Alpaca 1046 "Not connected" error code?

Best wishes, Peter

```
15:37:41.258 Camera                    Starting driver initialisation for ProgID: ASCOM.AlpacaDynamic1.Camera, Description: Camera Sky Simulator for ALPACA
15:37:41.273 [Lib] Camera 16143        Connecting to device: 127.0.0.1:11111 through URL: http://127.0.0.1:11111
15:37:41.273 Camera                    Alpaca client created successfully
15:37:41.273 Camera                    Completed initialisation
15:37:41.275 Connected Set             Connecting to device
15:37:41.275 [Lib] Connected 16143     Setting AlpacaCamera Connected to True
15:37:41.289 [Lib] Connected 16143     HTTP PUT - UriBase: 'api/v1/Camera/0/', Device URI: http://127.0.0.1:11111/api/v1/camera/0/connected
15:37:41.289 [Lib] Connected 16143     Sending form parameter Connected = True
15:37:41.289 [Lib] Connected 16143     Sending form parameter ClientID = 16143
15:37:41.289 [Lib] Connected 16143     Sending form parameter ClientTransactionID = 1
15:37:41.289 [Lib] Connected 16143     Sending header Accept = application/json
15:37:41.289 [Lib] Connected 16143     Sending header User-Agent = ASCOMAlpacaClient/2.0.8.0
15:37:41.289 [Lib] Connected 16143     Sending header Connection = keep-alive
15:37:41.289 [Lib] Connected 16143     Client Transaction ID: 1, Remote device: http://127.0.0.1:11111/, Full URI: http://127.0.0.1:11111/api/v1/camera/0/connected

15:37:41.406 [Lib] Connected 16143     Received header Content-Length = 113.
15:37:41.406 [Lib] Connected 16143     Received header Content-Type = application/json; charset=utf-8.
15:37:41.406 [Lib] Connected 16143     Returned data content type: 'application/json; charset=utf-8'
15:37:41.406 [Lib] Connected 16143     Received 113 bytes, ReadAsByteArrayAsync time: 0ms, Overall time: 117ms.
15:37:41.406 [Lib] Connected 16143     JSON response: {"ClientTransactionID":0,"ServerTransactionID":120,"ErrorNumber":1026,"ErrorMessage":"Value not set Connected"}

15:37:41.504 [Lib] Connected 16143     Received an Alpaca error: ValueNotSet. Error message: Value not set Connected
15:37:41.504 [Lib] Connected 16143     Received Alpaca error code: ValueNotSet (0x0402), the equivalent COM error HResult error code is -2147220478 (0x80040402)
15:37:41.504 [Lib] Connected 16143      Alpaca value not set error, throwing ValueNotSetException - ErrorMessage: "Value not set Connected", ErrorNumber: 0x80040402
```

[image]

Peter-Simpson, 2024-11-26:
Hi Han,

Having had a look at the driver code, it may be that you are expecting the Connected form parameter to be cased as "connected" rather than "Connected", I suspect that the error message is a result of your Alpaca device not finding the Connected parameter because it is making a cased search for "connected".

I've copied an extract from the [Alpaca API Reference](https://github.com/ASCOMInitiative/ASCOMRemote/raw/main/Documentation/ASCOM%20Alpaca%20API%20Reference.pdf) below 

If you want to be absolutely sure, I suggest that you make all Alpaca parameter name comparisons case insensitive.

Best wishes, Peter

[image]

han-k59, 2024-11-26:
Hi Peter,

My code expects for method connected the parameter "Connected".  I will see if making all uppercase or downcase makes a difference

Han

han-k59, 2024-11-26:
Probably best way forward is to analyse with a third program the TCP/IP communication.

han-k59, 2024-11-26:
I have installed Wireshark. Connected=true seems to be the parameter in the first PUT message.  But why is it always accepted for Ascom 6.6 and not for Ascom 7? 

I will investigate more tomorrow.

han-k59, 2024-11-27:
Thanks for the feedback.  Now it is clear I have to look into the TCP/IP communication. This is absolutely not my expertise. But I will see in the next days.

>>This means that under Platform 7 every client command may fail if a few tens of seconds have passed since the last successful >>command from the client.
Normally messages will be transferred continuously for the mount poisition. For other equipment there will be less comminication. Could this be a problem? Does the the ASCOM client automatically reconnect?

Cheers, Han

pchev, 2024-11-27:
Hi Peter and Han,

I take a look at the issue because I originally write this part of the Alpaca server.

The problem is a timeout when we read the body data from the socket after reading the header to get the value for Content-Length. This result in an empty body to processed.

The read timeout was set to 100ms, increasing it to 500ms fix the issue.
This is strange because I expected the client send the header and the body all at the same time, but it look like I am wrong.

The fix is to change line 320 of cu_tcpserver.pas :
body:=RecvBufferStr(cl,100)
by
body:=RecvBufferStr(cl,500)

The test for "connected" value is not an issue because the parameter search is made in a case insensitive way. 
But it is nicer if this line is modified in each device class ProcessPutRequest():
if GetParamBool(params,'Connected',ok) then

Patrick

Peter-Simpson, 2024-11-27:
Hi Patrick,

Thanks very much for finding a work-round for the issue! 

I don't know why the PUT is split across two packets, I'll investigate to see if I have any control over that aspect of the HTTPClient behaviour.

Best wishes, Peter

Peter-Simpson, 2024-11-28:
My investigation has found that the HttpClient behaviour of sending headers and body in separate packets for HTTP write operations is "wired in" at the code level and cannot be changed by application developers. 

The HttpClient is widely used but there are no reports of issues with the two packet behaviour, so it seems that server devices in general handle this behaviour without issue.

Since a work-round for the driver has been identified, thanks Patrick!, I am closing the issue because no change is required in the Platform.

Best wishes, Peter

closed 2024-11-28 by Peter-Simpson

pchev, 2024-11-28:
Thank you Peter to look at this.

Increasing the timeout is not a problem and has no implication on the performance because the function return as soon it read the specified number of bytes, independently of the timeout that is just a safeguard again a permanent lock.

han-k59, 2024-11-28:
Thanks both of you!  My application is now updated and works seamlessly with ASCOM 7. :)

Peter-Simpson, 2024-11-28:
Thanks for letting me know, I have been holding back the Platform 7 Update 2 release in case a Platform fix was required.