ASCOMInitiative/ASCOMPlatform #80: unable to copy installation data to local folders

Jones1403, 2025-03-15:
Hello,

I experience an issue installing platform 7 on a windows 11 PC with the following notice.
"Unable to copy installation data to local folders"

Attached you have the log files that were created in "My Documents"

The PC has no previous installations of ASCOM.

[ASCOMPlatform7Install113465245696.txt](https://github.com/user-attachments/files/19262176/ASCOMPlatform7Install113465245696.txt)

[ASCOM.SetProfileACL.1408.542090.txt](https://github.com/user-attachments/files/19262177/ASCOM.SetProfileACL.1408.542090.txt)

What can be the issue?

Jona

Peter-Simpson, 2025-03-15:
Hi Jona,

I don't recall seeing this issue reported before so I'm not sure of the cause. I see that you are using an "Enterprise" version of Windows, is your PC provided and supported by the IT department of a company or educational institution? If so, your account may not have sufficient permission and you may need to get help from your IT support team to get the Platform installed.

Your installer log tells me that you are not using the latest Platform 7 installer: [Platform 7.0 Update 2](https://github.com/ASCOMInitiative/ASCOMPlatform/releases/tag/v7.0.2). I suggest that you download and install this version instead of the initial release you are currently using.

If it still doesn't work, please create and post a detailed installer log as described [here ](https://ascomtalk.groups.io/g/Help/wiki/12870) and I'll see if it provides any indication as to why you are having this issue.

Best wishes, Peter

Jones1403, 2025-03-15:
Hi Peter,

Thanks for your fast response. My PC is not managed by the IT department of the university I work for. I have full adminstrator permissions. 

I initially downloaded the Update 2 directly from the ascom website twice. Yet, because I experienced this issue, I searched for an older release to test. The original platform 7 release however yields the same result.

I executed the directions from the groups.io message, the log file is attached. Yet, I don't believe it contains more information.

[ascomlog.txt](https://github.com/user-attachments/files/19265004/ascomlog.txt)

Jones1403, 2025-03-16:
Hello Peter,

The PC, although purchased by the university, was never at the IT department. I deliberately chose not to have it messed up by group policies and other restrictions from IT when I'm more than capable manage my own PC. I received it directly from the manufacturer and completed the installation myself. That is why I'm surprised by this message, especially since installing version 6.6 is possible without issue. 

Also I'm not running a special AV software except for windows defender.

Why would it be able to run version 6.6 SP2 with the same type of installer but not 7.02?

Peter-Simpson, 2025-03-17:
I can only hypothesise about why you experienced this issue, I upgrade and downgrade regularly without issues as part of testing and to support people who use earlier Platforms. It's possible that some files were left in the temp directory of your PC for some reason, which interfered with subsequent Platform 7 installation; these would likely have been removed by the successful installation of Platform 6.

If I understand your post correctly, you now have Platform 7 installed OK. Please can you confirm?

Peter-Simpson, 2025-03-18:
Thanks for confirming, I will close the issue.

Best wishes, Peter

closed 2025-03-18 by Peter-Simpson