ASCOMInitiative/ASCOMPlatform #82: ASCOM Device HUB malforms Alpaca canmoveaxis request on AlpacaDynamic1.Telescope

jdmorriso, 2025-07-12:
Issue:
ASCOM Device Hub sends the following malformed URL when trying to connect to an Alpaca Telescope
/api/v1/telescope/0/canmoveaxis??ClientID=50985&ClientTransactionID=37&Axis=0
(See ?? in URL)

Environment:
* Recently upgraded to ASCOM Platform 7 Update 2 running on Win11
* Alpaca Benro Polaris Driver v1.0.1 used

Steps to Reproduce:
* Discover Alpaca Telescope Device using ASCOM Driver Setup (works ok).
* Using ASCOM Device Hub
* Choose Telescope Device as ASCOM.AlpacaDynamic1.Telescope
* On the Telescope tab, click on Connect Telescope.

Log of Device Hub:
17:59:28.117: Telescope - Capabilities: Get CanFindHome: False
17:59:28.222: Telescope - Capabilities: Get CanPark: True
17:59:28.312: Telescope - Capabilities: Get CanPulseGuide: False
17:59:28.419: Telescope - Capabilities: Get CanSetDeclinationRate: False
17:59:28.510: Telescope - Capabilities: Get CanSetGuideRates: False
17:59:28.604: Telescope - Capabilities: Get CanSetPark: False
17:59:28.697: Telescope - Capabilities: Get CanSetPierSide: False
17:59:28.789: Telescope - Capabilities: Get CanSetRightAscensionRate: False
17:59:28.880: Telescope - Capabilities: Get CanSetTracking: True
17:59:28.972: Telescope - Capabilities: Get CanSlew: True
17:59:29.064: Telescope - Capabilities: Get CanSlewAltAz: True
17:59:29.159: Telescope - Capabilities: Get CanSlewAltAzAsync: True
17:59:29.252: Telescope - Capabilities: Get CanSlewAsync: True
17:59:29.371: Telescope - Capabilities: Get CanSync: True
17:59:29.463: Telescope - Capabilities: Get CanSyncAltAz: False
17:59:29.555: Telescope - Capabilities: Get CanUnpark: True
17:59:29.652: Telescope - Capabilities: Calling CanMoveAxis( Primary ) - (failed). Details follow:
17:59:29.652: Telescope - Capabilities: ASCOM.DriverException: Error calling method: CanMoveAxis, HTTP Completion Status: BadRequest, Error Message:
{"title": "Bad Alpaca Request", "description": "Missing, empty, or misspelled parameter \"ClientID\""} ---> System.Exception: Error calling method: CanMoveAxis, HTTP Completion Status: BadRequest, Error Message:
{"title": "Bad Alpaca Request", "description": "Missing, empty, or misspelled parameter \"ClientID\""}
   --- End of inner exception stack trace ---
   at ASCOM.DriverAccess.MemberFactory.CheckDotNetExceptions(String memberName, Exception e) in C:\ASCOM Build\Export\ASCOM.DriverAccess\MemberFactory.cs:line 665
   at ASCOM.DriverAccess.MemberFactory.MethodTargetInvocationExceptionHandler(String memberName, Exception e) in C:\ASCOM Build\Export\ASCOM.DriverAccess\MemberFactory.cs:line 713
   at ASCOM.DriverAccess.MemberFactory.CallMember(Int32 memberCode, String memberName, Type[] parameterTypes, Object[] parms) in C:\ASCOM Build\Export\ASCOM.DriverAccess\MemberFactory.cs:line 500
   at ASCOM.DriverAccess.Telescope.CanMoveAxis(TelescopeAxes Axis) in C:\ASCOM Build\Export\ASCOM.DriverAccess\Telescope.cs:line 267
   at ASCOM.DeviceHub.TelescopeService.CanMoveAxis(TelescopeAxes axis) in C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\Services Classes\Implementations\TelescopeService.cs:line 223
   at ASCOM.DeviceHub.TelescopeManager.CanMoveAxis(TelescopeAxes axis) in C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\DeviceManagers\TelescopeManagerAccess.cs:line 597
   at ASCOM.DeviceHub.TelescopeCapabilities..ctor(TelescopeManager mgr) in C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\Business Object Classes\Telescope Classes\TelescopeCapabilities.cs:line 43
   at ASCOM.DeviceHub.TelescopeManager.ReadInitialTelescopeDataTask() in C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\DeviceManagers\TelescopeManager.cs:line 1008
   at ASCOM.DeviceHub.TelescopeManager.<ConnectTelescope>b__241_0() in C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\DeviceManagers\TelescopeManager.cs:line 220
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
17:59:29.652: Telescope - Others:       Attempting to initialize telescope operation caught an un-handled exception. Details follow:
17:59:29.652: Telescope - Others:       ASCOM.DriverException: Error calling method: CanMoveAxis, HTTP Completion Status: BadRequest, Error Message:
{"title": "Bad Alpaca Request", "description": "Missing, empty, or misspelled parameter \"ClientID\""} ---> System.Exception: Error calling method: CanMoveAxis, HTTP Completion Status: BadRequest, Error Message:
{"title": "Bad Alpaca Request", "description": "Missing, empty, or misspelled parameter \"ClientID\""}
   --- End of inner exception stack trace ---
   at ASCOM.DriverAccess.MemberFactory.CheckDotNetExceptions(String memberName, Exception e) in C:\ASCOM Build\Export\ASCOM.DriverAccess\MemberFactory.cs:line 665
   at ASCOM.DriverAccess.MemberFactory.MethodTargetInvocationExceptionHandler(String memberName, Exception e) in C:\ASCOM Build\Export\ASCOM.DriverAccess\MemberFactory.cs:line 713
   at ASCOM.DriverAccess.MemberFactory.CallMember(Int32 memberCode, String memberName, Type[] parameterTypes, Object[] parms) in C:\ASCOM Build\Export\ASCOM.DriverAccess\MemberFactory.cs:line 500
   at ASCOM.DriverAccess.Telescope.CanMoveAxis(TelescopeAxes Axis) in C:\ASCOM Build\Export\ASCOM.DriverAccess\Telescope.cs:line 267
   at ASCOM.DeviceHub.TelescopeService.CanMoveAxis(TelescopeAxes axis) in C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\Services Classes\Implementations\TelescopeService.cs:line 223
   at ASCOM.DeviceHub.TelescopeManager.CanMoveAxis(TelescopeAxes axis) in C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\DeviceManagers\TelescopeManagerAccess.cs:line 597
   at ASCOM.DeviceHub.TelescopeCapabilities..ctor(TelescopeManager mgr) in C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\Business Object Classes\Telescope Classes\TelescopeCapabilities.cs:line 43
   at ASCOM.DeviceHub.TelescopeManager.ReadInitialTelescopeDataTask() in C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\DeviceManagers\TelescopeManager.cs:line 1008
   at ASCOM.DeviceHub.TelescopeManager.<ConnectTelescope>b__241_0() in C:\ASCOM Build\Export\Drivers and Simulators\ASCOM Device Hub\DeviceHub\DeviceManagers\TelescopeManager.cs:line 220
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
17:59:30.324: Telescope - Others:       Set Connected flag -> False (disconnected)
17:59:30.324: Telescope - Others:       Disconnection took 0 milliseconds.
17:59:30.327: Telescope - Others:       Get Connected flag - False
17:59:30.327: Telescope - Others:       Get Connected flag - False

Log of Alpaca Device:
2025-07-12T07:59:45.005 INFO 192.168.50.73 -> GET /api/v1/telescope/0/interfaceversion?ClientID=50985&ClientTransactionID=1
2025-07-12T07:59:45.006 INFO 192.168.50.73 <- 3
2025-07-12T07:59:46.337 INFO 192.168.50.73 -> GET /api/v1/telescope/0/canfindhome?ClientID=50985&ClientTransactionID=5
2025-07-12T07:59:46.337 INFO 192.168.50.73 <- False
2025-07-12T07:59:46.443 INFO 192.168.50.73 -> GET /api/v1/telescope/0/canpark?ClientID=50985&ClientTransactionID=7
2025-07-12T07:59:46.443 INFO 192.168.50.73 <- True
2025-07-12T07:59:46.823 INFO 192.168.50.73 -> GET /api/v1/telescope/0/cansetpark?ClientID=50985&ClientTransactionID=15
2025-07-12T07:59:46.823 INFO 192.168.50.73 <- False
2025-07-12T07:59:47.577 INFO 192.168.50.73 -> GET /api/v1/telescope/0/cansync?ClientID=50985&ClientTransactionID=31
2025-07-12T07:59:47.578 INFO 192.168.50.73 <- True
2025-07-12T07:59:47.682 INFO 192.168.50.73 -> GET /api/v1/telescope/0/cansyncaltaz?ClientID=50985&ClientTransactionID=33
2025-07-12T07:59:47.682 INFO 192.168.50.73 <- False
2025-07-12T07:59:47.774 INFO 192.168.50.73 -> GET /api/v1/telescope/0/canunpark?ClientID=50985&ClientTransactionID=35
2025-07-12T07:59:47.775 INFO 192.168.50.73 <- True
2025-07-12T07:59:47.866 INFO 192.168.50.73 -> GET /api/v1/telescope/0/canmoveaxis??ClientID=50985&ClientTransactionID=37&Axis=0
2025-07-12T07:59:47.985 INFO 192.168.50.73 -> PUT /api/v1/telescope/0/connected
2025-07-12T07:59:47.986 INFO 192.168.50.73 -> {'Connected': 'False', 'ClientID': '50985', 'ClientTransactionID': '39'}

jdmorriso, 2025-07-12:
Also send the malformed URL with axisrates eg.
GET /api/v1/telescope/0/axisrates??ClientID=18393&ClientTransactionID=47&Axis=1

Peter-Simpson, 2025-07-12:
Hi, 

This sounds like a known issue in the Dynamic Drivers that Platform 7 creates to talk to Alpaca devices rather than issue in Device Hub itself.

If you are using a Dynamic Client please try this updated Platform to see if it fixes the issues you are seeing.

[Platform 7 Update 3 RC 7](https://download.ascom-standards.org/beta/ASCOMPlatform703-rc.7.4754.exe)

Best wishes, Peter

Peter-Simpson, 2025-07-13:
Hi Dave,

ISSUE 3 - The original author took the approach of just providing a rounded "half way" value for each axis range advertised by the driver, hence you see 5 degrees/second when DH reads the axis rate range of 1 to 9. I could provide more drop-down values in each range e.g.
`Min, Mid, Max`
or 
`Min, Mid-Min, Mid, Mid-Max, Max`.
Would this work for you?

ISSUE 4 - I'll see if I can prevent IsPulseGuiding polling as suggested but DH is extremely complex under the hood and I have to be very careful of unintended consequences when working with its polling code.

Best wishes, Peter

jdmorriso, 2025-07-13:
Peter,
I think your proposal for issue 3 would be fine. Issue 3 is more important
than issue 4. I'm ok to close it if 4 cannot be done..

Best Regards
Dave

Peter-Simpson, 2025-07-15:
Great, thanks for letting me know Dave.

Best wishes, Peter

closed 2025-07-15 by Peter-Simpson